pyusb/pyusb #82: readme.md could mention that pip is supported

patrickmmartin, 2015-01-02:
I used  

    pushd c:\Python27\Scripts && pip install pyusb==1.0.0b2 && popd

to install - I saw the issue where pyusb with no version will no work until the version is out of beta, so please consider pluggin this approach as the installation method

walac, 2015-01-03:
I don't mention pip in the [README](https://github.com/walac/pyusb/blob/master/README.rst) file because it is supported in the contribution mode.

arve0, 2015-04-20:
Can you elaborate what you mean by 
> in the contribution mode.

Any reason why pip is not the recommended way of installation?

patrickmmartin, 2015-04-25:
well pip works, but at the time of testing, not without an explicit version no., which is not 100% obvious

walac, 2015-04-25:
By *contribution mode* I mean that people that don't fully work (and don't have obligation to) on the project kindly push releases to pip. Currently I cannot work on it, but there is a big demand for it, so I will see if I change this and push to pip whenever a new version is released.

commit 44bbbc33 (2015-06-14): Update docs to include pip install.

closes #82
closes #92
commit 39816be3 (2015-06-14): Update docs to include pip install.

closes #82
closes #92
closed 2015-06-14 by walac