pyusb/pyusb #83: libusb_strerror not implemented in libusb-1.0.9

stephenrauch, 2015-01-05:
Using pyusb 1.0.0b2, and libusb-1.0.9 on Linux

    File "/usr/lib/python2.7/site-packages/usb/core.py", line 364, in read
      return self.device.read(self, size_or_buffer, timeout)
    File "/usr/lib/python2.7/site-packages/usb/core.py", line 918, in read
      self.__get_timeout(timeout))
    File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 787, in intr_read
      timeout)
    File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 872, in __read
      _check(retval)
    File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 552, in _check
      raise USBError(_strerror(ret), ret, _libusb_errno[ret])
    File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 541, in _strerror
      return _lib.libusb_strerror(errcode).decode('utf8')
    File "/usr/lib/python2.7/ctypes/__init__.py", line 378, in __getattr__
      func = self.__getitem__(name)
    File "/usr/lib/python2.7/ctypes/__init__.py", line 383, in __getitem__
      func = self._FuncPtr((name_or_ordinal, self))
    AttributeError: /usr/lib/libusb-1.0.so.0: undefined symbol: libusb_strerror

Can't use libusb-1.0.19 because libudev is not readily available on my distribution.  I get the following when trying to build it: `configure: error: "udev support requested but libudev not installed"`.  Tried to go down the path of building libudev, but it had its own dependencies, so too many dragons down that path...

I have increased my timeout, and I am not hitting this now, but it probably isn't the best behavior to have the error reporting code take an exception.

As a hacky work around I changed _strerror from:

    def _strerror(errcode):
        return _lib.libusb_strerror(errcode).decode('utf8')

to:

    def _strerror(errcode):
        errcode_strings = [
            "Success",
            "Input/Output Error",
            "Invalid parameter",
            "Access denied (insufficient permissions)",
            "No such device (it may have been disconnected)",
            "Entity not found",
            "Resource busy",
            "Operation timed out",
            "Overflow",
            "Pipe error",
            "System call interrupted (perhaps due to signal)",
            "Insufficient memory",
            "Operation not supported or unimplemented on this platform",
            "Other error",
            ]

        try:
            return _lib.libusb_strerror(errcode).decode('utf8')

        except AttributeError:
            errcode_index = -errcode;
            if errcode_index < 0 or errcode_index >= len(errcode_strings):
                errcode_index = -1
            return errcode_strings[errcode_index]

walac, 2015-01-08:
Hi, I think issue #69 fixes that. Have you tried git master?

stephenrauch, 2015-01-08:
Splendid, this bug is indeed a duplicate of https://github.com/walac/pyusb/issues/69.  I have confirmed that https://github.com/walac/pyusb/commit/c46563c6b54acb9c2cbcbf7755f31fa3614e198b fixes the problem I reported here, and did so the day before I reported it.  I made the mistake of looking at open bugs, not thinking about the tip.

Note: I also updated the bug description above with instructions to build libusb-1.0.19 without libudev, which also solved my problem.

Thanks.

closed 2015-01-08 by stephenrauch