pyusb/pyusb #84: USBError(19, u'No such device ...) in libusb1.release_interface()

stephenrauch, 2015-01-08:
Seeing the following on the console on removing the usb device:

    Exception usb.core.USBError: USBError(19, u'No such device (it may have been disconnected)') in
    <bound method Device.__del__ of <DEVICE ID 0bd3:0333 on Bus 002 Address 066>> ignored

Using python 2.7.4, libusb-1.0.19, the current tip of pyusb (previously also occurred on beta 2) on Linux.

According to https://docs.python.org/2/reference/datamodel.html#object.__del__

> **Warning** Due to the precarious circumstances under which \__del__() methods are invoked, exceptions that occur during their execution are ignored, and a warning is printed to sys.stderr instead. Also, when \__del__() is invoked in response to a module being deleted (e.g., when execution of the program is done), other globals referenced by the \__del__() method may already have been deleted or in the process of being torn down (e.g. the import machinery shutting down). For this reason, \__del__() methods should do the absolute minimum needed to maintain external invariants. Starting with version 1.5, Python guarantees that globals whose name begins with a single underscore are deleted from their module before other globals are deleted; if no other references to such globals exist, this may help in assuring that imported modules are still available at the time when the \__del__() method is called.

In an effort to try and find what I was doing wrong I ended up changing core.Device._finalize_object() from:

    def _finalize_object(self):
        self._ctx.dispose(self)
        return 

to:

    def _finalize_object(self):
        try:
            self._ctx.dispose(self)
        except USBError, e:
            # [Errno 19] No such device (it may have been disconnected)
            if e.errno != 19:
                raise
            else:
                # for [Errno 19] dump full stack
                import traceback, sys

                trace = '----------------------\n'
                trace += 'Traceback (most recent call last):\n'
                for line in traceback.format_stack()[:-1]:
                    trace += line
                trace += '.\n'
                for line in traceback.format_tb(sys.exc_info()[2]):
                    trace += line
                trace += "%s: %s\n" % (e.__class__.__name__, e)
                trace += '----------------------\n'
                sys.stderr.write(trace)

This yielded the following:

    ...
        del usb_devs[usbid]
      File "/usr/lib/python2.7/site-packages/usb/_objfinalizer.py", line 84, in __del__
        self.finalize()
      File "/usr/lib/python2.7/site-packages/usb/_objfinalizer.py", line 155, in finalize
        self._do_finalize_object()
      File "/usr/lib/python2.7/site-packages/usb/_objfinalizer.py", line 71, in _do_finalize_object
        self._finalize_object()
    .
      File "/usr/lib/python2.7/site-packages/usb/core.py", line 1039, in _finalize_object
        self._ctx.dispose(self)
      File "/usr/lib/python2.7/site-packages/usb/core.py", line 226, in dispose
        self.release_all_interfaces(device)
      File "/usr/lib/python2.7/site-packages/usb/core.py", line 223, in release_all_interfaces
        self.managed_release_interface(device, i)
      File "/usr/lib/python2.7/site-packages/usb/core.py", line 160, in managed_release_interface
        self.backend.release_interface(self.handle, i)
      File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 799, in release_interface
        _check(self.lib.libusb_release_interface(dev_handle.handle, intf))
      File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 588, in _check
        raise USBError(_strerror(ret), ret, _libusb_errno[ret])
    USBError: [Errno 19] No such device (it may have been disconnected)

The exception is occurring in libusb1.release_interface(), so I assume is a behavior of the libusb library. I am unsure how my code could cause this. The exception happens every time when the device is in a certain configuration, but does not happen when the device is in another configuration.

My knowledge of the usb bus is minimal so maybe this is expected behavior, and the error code is legitimate.  However, the exception is being 'caught' by python because it is in the \__del__() method.  The resulting problem this presents in my application is the 'errant' message to stderr listed at the top of this comment.

walac, 2015-01-09:
My guess is that the kernel releases the resources allocated by `claim_interface` operation once the device is disconnected. If I am right, I think the only way to safely do the cleanup is after bug #64 lands, then at disconnect event, we can release the interface.

We can also try to reenumerate the device before call `libusb_release_interface` to check if it is still there (inside the backend), but it feels hacky.

walac, 2015-01-17:
What kernel and libusb version are you using? I tried to reproduce the problem but I couldn't. Could you please also elaborate more on how to reproduce this?

stephenrauch, 2015-01-18:
Kernel is Centos 5, which is 2.6.18.  I am currently using Libusb 2.0.19.  But as I stated earlier:

> The exception happens every time when the device is in a certain configuration, but does
> not happen when the device is in another configuration.

This may be difficult for you to reproduce, since you don't have the device, and in my case it was specific to a certain configuration on that device.

My suggestion would be to catch and ignore the error in release_interface.  If you are confident that your code has a correct device (it worked earlier) and owns the interface that it is attempting to release, then the code should be fine, and an error of "No Such Device" could be confidently ignored.

------------

I can and will test anything you need.  Let me know.

walac, 2015-01-18:
Could you please try this patch and check if the error is gone?

```patch
From 8ab3b5b991b859aaf4e2b6d1813fbdbadfb50d5d Mon Sep 17 00:00:00 2001
From: Wander Lairson Costa <wander.lairson@gmail.com>
Date: Sun, 18 Jan 2015 10:30:48 -0200
Subject: [PATCH] core: Handle exceptions when releasing the interfaces.

At object destruction, claimed interfaces are released. If the device
is disconnected, the call to release_interface may fail and raise an
exception.

closes #84.
---
 usb/core.py | 7 ++++++-
 1 file changed, 6 insertions(+), 1 deletion(-)

diff --git a/usb/core.py b/usb/core.py
index 6174e93..50446db 100644
--- a/usb/core.py
+++ b/usb/core.py
@@ -220,7 +220,12 @@ class _ResourceManager(object):
     def release_all_interfaces(self, device):
         claimed = copy.copy(self._claimed_intf)
         for i in claimed:
-            self.managed_release_interface(device, i)
+            try:
+                self.managed_release_interface(device, i)
+            except USBError:
+                # Ignore errors when releasing the interfaces
+                # When the device is disconnected, the call may fail
+                pass
 
     def dispose(self, device, close_handle = True):
         self.release_all_interfaces(device)
-- 
1.9.1
```

stephenrauch, 2015-01-20:
Verified this patch prevents the exception from getting back up to the \__del__() method and subsequently being reported via stderr.

walac, 2015-01-20:
Thanks for testing the patch.

commit 5e9ef153 (2015-01-20): core: Handle exceptions when releasing the interfaces.

At object destruction, claimed interfaces are released. If the device
is disconnected, the call to release_interface may fail and raise an
exception.

closes #84.
closed 2015-01-20 by walac