pyusb/pyusb #85: iso transfer callback incorrectly raise exception in callback thread which does not interrupt main thread

QianJin2013, 2015-01-15:
In iso mode, there are a __callback function, when transfer data, if i suddenly disconnect USB device, 
it will get an exception as below,   but this does not cause the caller of ep_in.read(buf) raise exception because the caller is in main thread, the callback seems in other thread, as a result, my program will still in loop of  _check(_lib.libusb_handle_events(ctx)) in submit function.

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/Library/Python/2.7/site-packages/usb/backend/libusb1.py", line 689, in __callback
    _transfer_errno[status])
usb.core.USBError: [Errno 35] Transfer was cancelled

<b>So, i changed libusb1.py:</b>

    def submit(self, ctx = None):
        self.__callback_done = 0
        self.__callback_err = 0      #this is what i added
        _check(_lib.libusb_submit_transfer(self.transfer))

        while not self.__callback_done and not self.__callback_err:
            _check(_lib.libusb_handle_events(ctx))

        if self.__callback_err:     #this is what i added
            raise usb.USBError(_str_transfer_error[self.status],
                               self.status,
                               _transfer_errno[self.status])

        return self.__compute_size_transf_data()

    def __compute_size_transf_data(self):
        return sum([t.actual_length for t in
                    _get_iso_packet_list(self.transfer.contents)])

    def __set_packets_length(self, n, packet_length):
        _lib.libusb_set_iso_packet_lengths(self.transfer, packet_length)
        r = n % packet_length
        if r:
            iso_packets = _get_iso_packet_list(self.transfer.contents)
            iso_packets[-1].length = r

    def __callback(self, transfer):
        self.status = int(transfer.contents.status)
        if transfer.contents.status == LIBUSB_TRANSFER_COMPLETED:
            self.__callback_done = 1
        else:
            self.__callback_err = 1
            # raise usb.USBError(_str_transfer_error[status],        #this is what i deleted
            #                    status,
            #                    _transfer_errno[status])

walac, 2015-01-16:
hrm I think it is better mark the error in the callback is raise the exception after completion.

QianJin2013, 2015-01-16:
thank you for you review. I mean that the ep_in.read(in turn call iso_read -> handler.submit) will never  end nor timeout nor raise exception if device is disconnected when reading. Do you think this is expected behavior?

walac, 2015-01-17:
Fixed. Thanks for the bug report.

closed 2015-01-17 by walac

QianJin2013, 2015-01-18:
Thank you very much.

commit ce019936 (2015-01-17): libusb1: fix hanging when the device is disconnected in a iso transfer.

If the device is disconnected when a isochronous transfer is ongoing,
it might raise an exception in the callback and the 'callback done'
signaling flag is never set. This causes the program to hang
forever.

We move the error handling to the caller function, after the callback
returns. In this way, the exception is raised in the main thread and
client software is notified about the exceptional situation.

closes #85.