pyusb/pyusb #86: NotImplementedError(_strerror(ret)) for Libusb 1.0

t-kopp, 2015-01-15:
Hi,
when trying to use pyusb with libusb 1.0 on Windows 7 machine I'm running into this error:

>Traceback (most recent call last):
>  File "<stdin>", line 2, in <module>
>  File "usb\core.py", line 693, in __str__
>    configuration = self.get_active_configuration()
>  File "usb\core.py", line 805, in get_active_configuration
>    return self._ctx.get_active_configuration(self)
>  File "usb\core.py", line 208, in get_active_configuration
>    self.managed_open()
>  File "usb\core.py", line 105, in managed_open
>    self.handle = self.backend.open_device(self.dev)
>  File "usb\backend\libusb1.py", line 722, in open_device
>    return _DeviceHandle(dev)
>  File "usb\backend\libusb1.py", line 600, in __init__
>    _check(_lib.libusb_open(self.devid, byref(self.handle)))
>  File "usb\backend\libusb1.py", line 550, in _check
>    raise NotImplementedError(_strerror(ret))
>NotImplementedError: Operation not supported or unimplemented on this platform

script I'm trying to run:

```python
import usb.core
import usb.backend.libusb1

backend = usb.backend.libusb1.get_backend(find_library=lambda x: "libusb-1.0.dll")
dev     = usb.core.find(find_all=True, backend=backend)
for d in dev:
  print(d)
```

It does find the libusb dll - backend is a valid object.
This was reproduced on a 32-bit Win 7 machine with Python 2.7.9 32-bit, and libusb from master here: https://github.com/libusb/libusb and also with the latest Release 1.019. 

When trying to reproduce this on a x64 machine (everything else identical, python 32-bit 2.7.9, libusb from master and 1.019). It completely froze the cmd. No error message but unresponsive cmd - not even killable.

Any ideas on this?

walac, 2015-01-16:
This seems a problem with libusb installation. Did you use [zadig](http://zadig.akeo.ie/) to install the driver for your device?

t-kopp, 2015-01-18:
For now I only want to list the available devices. This should work regardless of whether every connected device has a libusb driver, right?

When using try, except around the print statement it actually works to print all the connected devices. I was able to narrow down the failing to these 4:

>==================================================
>Device Name       : 0000.001d.0000.001.008.000.000.000.000
>Description       : Dell ControlVault w/ Fingerprint Swipe Sensor
>Device Type       : Application Specific
>Connected         : Yes
>Safe To Unplug    : Yes
>Disabled          : No
>USB Hub           : No
>Drive Letter      : 
>Serial Number     : 
>Created Date      : 09.01.2015 11:37:41
>Last Plug/Unplug Date: 18.01.2015 11:27:15
>VendorID          : 0a5c
>ProductID         : 5801
>Firmware Revision : 1.01
>USB Class         : fe
>USB SubClass      : 00
>USB Protocol      : 00
>Hub / Port        : 
>Computer Name     : 
>Vendor Name       : 
>Product Name      : 
>ParentId Prefix   : 
>Service Name      : cvusbdrv
>Service Description: Dell ControlVault
>Driver Filename   : cvusbdrv.sys
>Device Class      : CVAULT
>Device Mfg        : Dell
>Power             : 
>USB Version       : 
>Driver Description: Dell ControlVault w/ Fingerprint Swipe Sensor
>Driver Version    : 6.0.6000.10408
>Instance ID       : USB\VID_0A5C&PID_5801&MI_00\7&1eb0f4e8&0&0000
>==================================================
>
>==================================================
>Device Name       : 0000.001d.0000.001.008.000.000.000.000
>Description       : Broadcom Usbccid Smartcard Reader (WUDF)
>Device Type       : Smart Card
>Connected         : Yes
>Safe To Unplug    : No
>Disabled          : No
>USB Hub           : No
>Drive Letter      : 
>Serial Number     : 
>Created Date      : 09.01.2015 11:37:41
>Last Plug/Unplug Date: 18.01.2015 20:56:29
>VendorID          : 0a5c
>ProductID         : 5801
>Firmware Revision : 1.01
>USB Class         : 0b
>USB SubClass      : 00
>USB Protocol      : 00
>Hub / Port        : 
>Computer Name     : 
>Vendor Name       : 
>Product Name      : 
>ParentId Prefix   : 8&21e2430a&0
>Service Name      : WUDFRd
>Service Description: 
>Driver Filename   : WUDFRd.sys
>Device Class      : SmartCardReader
>Device Mfg        : Broadcom
>Power             : 
>USB Version       : 
>Driver Description: Broadcom Usbccid Smartcard Reader (WUDF)
>Driver Version    : 15.44.59.312
>Instance ID       : USB\VID_0A5C&PID_5801&MI_01\7&1eb0f4e8&0&0001
>==================================================
>
>==================================================
>Device Name       : DW375 Bluetooth Module
>Description       : Dell Wireless 375 Bluetooth Module
>Device Type       : Unknown
>Connected         : Yes
>Safe To Unplug    : Yes
>Disabled          : No
>USB Hub           : No
>Drive Letter      : 
>Serial Number     : 446D57CE0B10
>Created Date      : 09.01.2015 11:37:41
>Last Plug/Unplug Date: 18.01.2015 20:38:30
>VendorID          : 413c
>ProductID         : 8187
>Firmware Revision : 5.17
>USB Class         : 00
>USB SubClass      : 00
>USB Protocol      : 00
>Hub / Port        : 
>Computer Name     : 
>Vendor Name       : 
>Product Name      : 
>ParentId Prefix   : 7&1e0c1e4f&0
>Service Name      : BTHUSB
>Service Description: Bluetooth Radio USB Driver
>Driver Filename   : BTHUSB.sys
>Device Class      : Bluetooth
>Device Mfg        : Broadcom
>Power             : 100 mA
>USB Version       : 2.00
>Driver Description: Dell Wireless 375 Bluetooth Module
>Driver Version    : 6.5.1.1201
>Instance ID       : USB\VID_413C&PID_8187\446D57CE0B10
>==================================================
>
>==================================================
>Device Name       : Port_#0001.Hub_#0003
>Description       : Generic USB Hub
>Device Type       : 
>Connected         : Yes
>Safe To Unplug    : Yes
>Disabled          : No
>USB Hub           : Yes
>Drive Letter      : 
>Serial Number     : 
>Created Date      : 09.01.2015 11:37:41
>Last Plug/Unplug Date: 18.01.2015 11:27:15
>VendorID          : 413c
>ProductID         : 2513
>Firmware Revision : 0.00
>USB Class         : 09
>USB SubClass      : 00
>USB Protocol      : 01
>Hub / Port        : Hub 2, Port 1
>Computer Name     : 
>Vendor Name       : 
>Product Name      : 
>ParentId Prefix   : 7&1d956223&0
>Service Name      : usbhub
>Service Description: Microsoft USB Standard Hub Driver
>Driver Filename   : usbhub.sys
>Device Class      : USB
>Device Mfg        : (Generic USB Hub)
>Power             : 
>USB Version       : 
>Driver Description: Generic USB Hub
>Driver Version    : 6.1.7601.18328
>Instance ID       : USB\VID_413C&PID_2513\6&7a7b691&0&1
>==================================================

Nothing sticks out to me - especially since there's a second port on the last device which is identical apart from the parentId Prefix.

walac, 2015-01-20:
You are trying to get all devices available, and, among them, there is at least one hub device (root hub). libusb 1.0 under Windows [does not have good support for hubs](https://github.com/libusb/libusb/blob/master/libusb/os/windows_usb.c#L2589), that's why you are getting this error.

closed 2015-01-20 by walac