pyusb/pyusb #89: Handle memory leak if release_all_interfaces throws exception

landron, 2015-01-28:
The problem is that exceptions are allowed in `dispose`, so the following code will leak a handle if the first line throws:

    def dispose(self, device, close_handle = True):
        self.release_all_interfaces(device)
        if close_handle:
            self.managed_close()

If my USB device is gone (re-enumeration) before dispose, then `release_interface` (in libusb0.py) will throw and the handle  won't be close. I do this in a loop and if I catch the exception, then I can see the handles number increasing (Process Explorer) with each pass through until ... 255 when the USB device (warning icon in "Device manager") refuses to communicate anymore until I close the script.

I hope I didn't miss something and I'll be happy to offer further informations and clarifications, if needed. Thank you for the great work!

walac, 2015-01-28:
hrm I guess issue #84 fixed that. Could you please make a test?

landron, 2015-01-28:
Apparently it works ... except when I call `usb.util.dispose_resources` (at the end of each loop) on the device and I see:

    Exception ignored in: <bound method Device.__del__ of <DEVICE ID 082b:100a on Bus 000 Address 002>>
    Traceback (most recent call last):
      File "..\usb\core.py", line 1035, in __del__
        self._ctx.dispose(self)
      File "..\usb\core.py", line 231, in dispose
        self.release_all_interfaces(device)
      File "..\usb\core.py", line 224, in release_all_interfaces
        self.managed_release_interface(device, i)
      File "..\usb\core.py", line 159, in managed_release_interface
        self.backend.release_interface(self.handle, i)
      File "..\usb\backend\libusb0.py", line 473, in release_interface
        _check(_lib.usb_release_interface(dev_handle, intf))
    OSError: exception: access violation reading 0x00000018

I'm a little bit lost because I don't know who catches this and why I didn't see it before, but that's beyond my expertise, I guess :D. But, even with this message, there are __no handle leaks__ anymore (`usb.util.dispose_resources` or the garbage collector, every 20 cycles or so).

On the other side, I need this `usb.util.dispose_resources` for the case when the device stays there, before the next loop.

walac, 2015-01-28:
Could you please post a test case?

landron, 2015-01-28:
It is a test case for our device. Sorry, I cannot show you the code (I can ask my manager, but I don't think she will agree), but I can provide you more information or try to reproduce it at home, in week-end with ... some another device. 

     while True:
	     # 1. get our device through libusb0.1 
	     # 2. some communication with the device
	     #   let the device in a state waiting for us to take some bulk data
	     time.sleep(8)
	     # 	 during this time the device reaches some timeout and reboots itself (re-enumeration - "that sound" in Device Manager)
	     usb.util.dispose_resources((device)

BTW, I don't have the "OSError: exception: access violation reading 0x00000018" with libusb1.0 and the WinUSB driver - it works just fine. I'll try to find the source of that exception these days.

commit 256981f3 (2015-02-01): core: Fix seg fault when dispose is called after device disconnection.

When dispose is called after the device is diconnected,
release_interface might fail (see commit 5e9ef1536), then the interface
stays in the claimed interfaces set. When the device is destroyed, its
__del__ method will call dispose again and, since the interface is
still marked as claimed, a new call to release_interface will be made
with an invalid device handle, causing a segmentation fault.

We fix that by always removing the interface from the claimed set.

closes #89.
closed 2015-02-01 by walac