pyusb/pyusb #95: How to detect if no backend is installed

hgrecco, 2015-03-18:
Is there any way to detect that no backend is installed? Right now I am doing
```python
try:
   d = usb.core.find()
except ValueError as e:
   if str(e) == 'No backend available':
       # do something
```

which is not nice in many ways. It would be really useful to raise a specific error.

walac, 2015-03-23:
Could you please elaborate more on the use case for when the backend is not available? Ex: user message, fall through to a custom backend, etc.

hgrecco, 2015-03-24:
I tell you my use case but I am sure there are othes. 

In [pyvisa-py](https://github.com/hgrecco/pyvisa-py), we use different packages (pyserial, pyusb, linux-gpib) to provide access to different type of instruments. To avoid forcing all of them into users that might need only one, we try to load them only at user request. If it is not available, we provide a nice error message ("Please install pyusb to use this feature", or similar). 

But having pyusb does not mean having a **usable** pyusb. I need to make sure that there is at least one backend and that is working.

commit b4b6ab44 (2015-06-14): Create a specialized class for "no backend" exception.

Now when no backend is found, the exception NoBackendError is raised,
allowing the application to handle it properly.

closes #95.
closed 2015-06-14 by walac