pyusb/pyusb #101: PyUSB compatible issue

FusionLi, 2015-05-20:
Hi, I tried the testing script for mouse on raspberry pi 2from http://www.orangecoat.com/how-to/read-and-decode-data-from-your-mouse-using-this-pyusb-hack but it gave me the error:
Traceback (most recent call last):
File "mouse.py", line 14, in
if dev.is_kernel_driver_active(interface) is True:
File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 992, in is_kernel_driver_active
interface)
File "/usr/local/lib/python2.7/dist-packages/usb/backend/init.py", line 363, in is_kernel_driver_active
not_implemented(self.is_kernel_driver_active)
File "/usr/local/lib/python2.7/dist-packages/usb/backend/__init_.py", line 79, in not_implemented
raise NotImplementedError(func.__name_)
NotImplementedError: is_kernel_driver_active

And then I solved this problem via install pyusb1.0.0b1 from https://github.com/arvydas/blinkstick-python/wiki/NotImplementedError-is_kernel_driver_active-on-some-Linux-systems. However, this package is not compatible with the one used by pyvisa because after I installed pyusb1.0.0b1, I cannot even use "open_resource() and list_resources()" methods in PyVISA package.

hgrecco, 2015-05-20:
It seems that PyUSB 1.0b2 uses some methods that are not supported by older libusb version and that are not available in all systems. I am not sure how important are these methods (for example maybe a failed call to `libusb_strerror` can be replaced by a generic error message indicating that getting the actual string is not possible with this libusb version)

walac, 2015-05-21:
Hi, if you try to pull from git, this `libusb_strerror` must be fixed.

FusionLi, 2015-05-21:
Thanks a lot, I solved it by installing from the git!

closed 2015-05-21 by walac