pyusb/pyusb #102: Pipe error with controlling Tektronix MSO4034

FusionLi, 2015-05-28:
I got a very strange error in controlling the Tektronix oscilloscope via PyVISA. Right now, I can only run six lines of command after getting it connected with raspberry pi 2. What I mean connected is that after I write the command "inst = rm.open_resource(rm.list_resources()[0])" in python interrupter. After whatever six lines of command, the pipe error appears in the following:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 187, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 165, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 281, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 32] Pipe error
```
I have to re-enable the USB connection on the signal oscilloscope to get it working again.

walac, 2015-05-31:
This seems to be a device related issue, PyVISA guys might help you better on this.

walac, 2015-06-14:
I am closing this issue, if you have farther info, feel free to reopen it.

closed 2015-06-14 by walac