pyusb/pyusb #106: Random NotImplementedError

jeremyherbert, 2015-07-07:
Hi,

I'm using PyUSB via the python-usbtmc library, but this bug looks like it is in pyusb. I have some very simple code which triggers the problem:

```
for i in range(100000):
    sc = usbtmc.Instrument(oscilloscope_address)
    try:
        data = sc.ask("*IDN?")
    finally:
        sc.release()
```

This code generally works about 150 times in a row, and then I get the following error:

```
---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
<ipython-input-2-5bf616a5821c> in <module>()
      1 for i in range(100000):
----> 2     sc = usbtmc.Instrument(oscilloscope_address)
      3     try:
      4         data = sc.ask("*IDN?")
      5     finally:

/home/jeremy/anaconda3/lib/python3.4/site-packages/python_usbtmc-0.6-py3.4.egg/usbtmc/usbtmc.py in __init__(self, *args, **kwargs)
    248         # initialize device
    249         if os.name == 'posix':
--> 250             if self.device.is_kernel_driver_active(0):
    251                 self.device.detach_kernel_driver(0)
    252 

/home/jeremy/anaconda3/lib/python3.4/site-packages/usb/core.py in is_kernel_driver_active(self, interface)
   1035         return self._ctx.backend.is_kernel_driver_active(
   1036                 self._ctx.handle,
-> 1037                 interface)
   1038 
   1039     def detach_kernel_driver(self, interface):

/home/jeremy/anaconda3/lib/python3.4/site-packages/usb/backend/__init__.py in is_kernel_driver_active(self, dev_handle, intf)
    363         and the backend will be unable to perform I/O.
    364         """
--> 365         _not_implemented(self.is_kernel_driver_active)
    366 
    367     def detach_kernel_driver(self, dev_handle, intf):

/home/jeremy/anaconda3/lib/python3.4/site-packages/usb/backend/__init__.py in _not_implemented(func)
     79 
     80 def _not_implemented(func):
---> 81     raise NotImplementedError(func.__name__)
     82 
     83 class IBackend(_objfinalizer.AutoFinalizedObject):

NotImplementedError: is_kernel_driver_active
```

If I try to run the code again (this is running in ipython notebook) I get a segfault in libusb:

```
python3[20464]: segfault at 208 ip 00007f5ea955d396 sp 00007ffd1e553ec0 error 4 in libusb-1.0.so.0.1.0[7f5ea9557000+17000]
```

Any idea how I can debug this further?

walac, 2015-07-26:
I am so sorry it took so much time to anwser, could you please provide the Operating System and kernel version you are using?

jeremyherbert, 2015-07-27:
`cat /proc/version` gives:
```
Linux version 3.19.0-16-generic (buildd@komainu) (gcc version 4.9.2 (Ubuntu 4.9.2-10ubuntu13) ) #16-Ubuntu SMP Thu Apr 30 16:09:58 UTC 2015
```

I am running Ubuntu 15.04.

walac, 2015-08-20:
Do you have more than one PyUSB version installed? Could you please run with `export PYUSB_DEBUG=debug` env variable and post the results?

mcuee, 2018-08-07:
Closing old issues since the OP has not come back for a long time. Please reopen if the issue still exists in latest version.

closed 2018-08-07 by mcuee