pyusb/pyusb #108: devnum is None

marian2015, 2015-08-06:
Hi,
in pyusb-1.0.0-0.10.b1.fc21 (Fedora 21) is devnum None. I upgraded from 0.4.3. 
usb.legacy and usb.find have the same problem.

walac, 2015-08-20:
I didn't understand your issue, could you please elaborate more?

marian2015, 2015-08-24:
Test code:
```python
#import usb
import usb.legacy as usb

for bus in usb.busses():
	for dev in bus.devices:
		print dev.devnum
```
With pyusb-0.4.3:
```
3
2
1
4
3
2
1
```
With pyusb-1.0.0:
```
None
None
None
None
None
None
None
```

BinaryMan32, 2016-08-12:
I just ran across this same issue after upgrading to ubuntu 16.04 which switched python-usb from 0.x to 1.x.

I think I have a fix for Bus.location and Device.devnum, can someone take a look?
https://github.com/walac/pyusb/pull/152

walac, 2016-08-12:
Merged, tks

closed 2016-08-12 by walac