pyusb/pyusb #114: Regression? NotImplementedError: is_kernel_driver_active

kphretiq, 2015-09-23:
System is raspberry pi b2, raspbian (Debian Wheezy armhf  7.8)

Error thrown when calling is_kernel_driver_active(0) when using version 1.0.0b2.

Downgrade to v1.0.0b1 seems to work just fine.

```bash
Traceback (most recent call last):
  File "./usb-demo.py", line 27, in <module>
    card_reader = CardReader()
  File "./usb-demo.py", line 20, in __init__
    if self.device.is_kernel_driver_active(0):
  File "/home/pi/pi-cardreader/venv/local/lib/python2.7/site-packages/usb/core.py", line 992, in is_kernel_driver_active
    interface)
  File "/home/pi/pi-cardreader/venv/local/lib/python2.7/site-packages/usb/backend/__init__.py", line 363, in is_kernel_driver_active
    _not_implemented(self.is_kernel_driver_active)
  File "/home/pi/pi-cardreader/venv/local/lib/python2.7/site-packages/usb/backend/__init__.py", line 79, in _not_implemented
    raise NotImplementedError(func.__name__)
```

Thanks!

walac, 2015-09-24:
Did you try git HEAD? Also, could you please run your app with `export PYUSB_DEBUG=debug` and post the logs?

mcuee, 2018-08-07:
Closing old issues since the OP has not come back for a long time. Please reopen if the issue still exists in latest version.

closed 2018-08-07 by mcuee