pyusb/pyusb #115: ValueError: No backend available

seamusdemora, 2015-10-03:
OS X 10.10.5, installed from Homebrew

Run code at https://github.com/walac/pyusb/blob/master/docs/tutorial.rst

ver:
2.7.10 (default, Sep 23 2015, 04:34:14) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.72)] 

output & error:
Traceback (most recent call last):
  File "/Users/jmoore/Desktop/Python/pyusb_ex1.py", line 5, in <module>
    dev = usb.core.find(idVendor=0xfffe, idProduct=0x0001)
  File "/usr/local/lib/python2.7/site-packages/usb/core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError: No backend available

haogefeifei, 2015-10-04:
same..

➜  ~  python
Python 2.7.10 (default, Aug 22 2015, 20:33:39) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> dev = usb.core.find()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Python/2.7/site-packages/usb/core.py", line 1221, in find
    raise ValueError('No backend available')
ValueError: No backend available

walac, 2015-10-05:
Please, define the environment variable `PYUSB_DEBUG=debug`and post the logs.

seamusdemora, 2015-10-05:
which logfile? I mean, where is it located?

walac, 2015-10-06:
Look in the `What's wrong` section in the [tutorial](https://github.com/walac/pyusb/blob/master/docs/tutorial.rst).

seamusdemora, 2015-10-07:
I'm sorry - that logfile isn't in my system. I give up - hope you get this fixed.

michaelllh, 2015-11-24:
Sorry，I don‘t understand why we need to post logs

walac, 2015-12-19:
Logs are necessary so I can figure out why the library is not loaded. I just pushed a new [FAQ](https://github.com/walac/pyusb/blob/master/docs/faq.rst) that might help.

michaelllh, 2015-12-21:
Thank you for you reply.I will try posed  the logs.

发自网易邮箱大师
在2015年12月20日 06:59，Wander Lairson Costa 写道:

Logs are necessary so I can figure out why the library is not loaded. I just pushed a new FAQ that might help.

tezro, 2016-02-11:
Having the same issue.
```
(venv) A:examples a$ python tagtool.py show
[nfc.clf] searching for reader on path usb
2016-02-11 13:21:48,799 ERROR:usb.libloader:'Libusb 1' could not be found
[usb.libloader] 'Libusb 1' could not be found
2016-02-11 13:21:48,799 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
[usb.backend.libusb1] Error loading libusb 1.0 backend
2016-02-11 13:21:48,800 ERROR:usb.libloader:'OpenUSB library' could not be found
[usb.libloader] 'OpenUSB library' could not be found
2016-02-11 13:21:48,800 ERROR:usb.backend.openusb:Error loading OpenUSB backend
[usb.backend.openusb] Error loading OpenUSB backend
2016-02-11 13:21:48,800 ERROR:usb.libloader:'Libusb 0' could not be found
[usb.libloader] 'Libusb 0' could not be found
2016-02-11 13:21:48,800 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
[usb.backend.libusb0] Error loading libusb 0.1 backend
```

kvoutyras, 2017-08-02:
Sorry for posting twice, I am trying to manually assign the library, but I cannot seem to do it.
The error I am getting is: 
`2017-08-02 13:01:12,941 ERROR:usb.libloader:'Libusb 1' could not be found`
`2017-08-02 13:01:12,942 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend`
`2017-08-02 13:01:12,946 ERROR:usb.libloader:'Libusb 1' could not be found`
`2017-08-02 13:01:12,946 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend`
`2017-08-02 13:01:12,947 ERROR:usb.libloader:'OpenUSB library' could not be found`
`2017-08-02 13:01:12,947 ERROR:usb.backend.openusb:Error loading OpenUSB backend`
`2017-08-02 13:01:12,949 ERROR:usb.libloader:'Libusb 0' could not be found`
`2017-08-02 13:01:12,950 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend`

 Furthermore, the tutorial page [https://github.com/walac/pyusb/blob/master/docs/faq.rst#how-can-i-pass-the-libusb-library-path-to-the-backend] is missing.

(I am using Windows 10-64 bit, Python3.6.1 pyusb 1.0.0)

walac, 2017-08-02:
@kvoutyras I fixed the link, thanks. Please, see my [comment](https://github.com/walac/pyusb/issues/120#issuecomment-319656652) at issue #120

mcuee, 2018-09-19:
All in all, not really a pyusb issue. I will try to create a FAQ section for the wiki page.

closed 2018-09-19 by mcuee

sanketdamame, 2019-09-21:
use conda for installing Pyusb. it installs all dependencies and things work fine.