pyusb/pyusb #118: Cannot import pyusb by pyInstaller3

xiekuncn, 2015-11-23:
when I package the a very simple pyusb code by the pyinstaller3. the code is like this.
```python
#!/bin/python2.7
# filename: usb-hook-test.py
import usb

print 'pyusb.version:{}'.format(usb.__version__)
dev = usb.core.find()
print dev
```

And I got a issue as:
```python
$ export PYUSB_DEBUG=debug
$ ./usb-hook-test/dist/usb-hook-test/usb-hook-test 
pyusb.version:1.0.0b2
2015-11-20 10:12:30,858 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/usb/backend/libusb1.py", line 879, in get_backend
  File "<string>", line 55, in _load_library
OSError: USB library could not be found
2015-11-20 10:12:30,858 ERROR:usb.backend.openusb:Error loading OpenUSB backend
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/usb/backend/openusb.py", line 736, in get_backend
  File "<string>", line 55, in _load_library
OSError: USB library could not be found
2015-11-20 10:12:30,859 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/usb/backend/libusb0.py", line 570, in get_backend
  File "<string>", line 55, in _load_library
OSError: USB library could not be found
Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "build/bdist.linux-x86_64/egg/usb/core.py", line 1199, in find
ValueError: No backend available
usb-hook-test returned -1
```
I had report this issue to the pyinstaller3 author, and the pyinstaller guys want to know the __"all the internals of pyusb load-engine"__, the detail of the issue is here.
https://github.com/pyinstaller/pyinstaller/issues/1682#issuecomment-158256874

Could you help us for this issue? 
Thanks a lot.

walac, 2015-12-19:
Hi, PyUSB uses [find_library](https://docs.python.org/2/library/ctypes.html) by default (the user can define his own).

mcuee, 2018-08-07:
Close this for now as the OP does not come back so I assume the answer is okay.

closed 2018-08-07 by mcuee