pyusb/pyusb #120: No backend available

michaelllh, 2015-11-24:
Hello,when I use pyusb.It occured a error.I don't know how to solve it.
All the error :
dev = usb.core.find(idVendor=0x0416,idProduct=0x5011)
File "/usr/lib/python2.7/site-packages/usb/core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError:No backend available

Kguruswami, 2015-12-03:
I am seeing the same error. unable to fix it :( 

>>> import usb
>>> import usb.core
>>> import usb.util
>>> dev = usb.core.find(idVendor=0x0955, idProduct=0xcf0b)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\usb\core.py", line 846, in find
    raise ValueError('No backend available')
ValueError: No backend available

toabi, 2015-12-08:
Getting this when trying to use it on a Synology DiskStation. Somehow there `ctypes.util.find_library` returns None for anything.

RenegadeTech, 2015-12-16:
I would also like to bump this issue.

toabi, 2015-12-16:
Is there any easy way to specify/hardcode where libraries are lying, instead of relying on the mechanisms to locate them?

walac, 2015-12-19:
Apologizes for the long delay, I am treating a disease and had to dedicate my time to recovery. I just pushed a [FAQ](https://github.com/walac/pyusb/blob/master/docs/faq.rst) which intends to address common issues. It is a WIP doc.

endolith, 2016-01-12:
[The documentation](https://github.com/walac/pyusb/blob/master/README.rst) is not very clear on this.

> Remember that you need libusb (1.0 or 0.1) or OpenUSB running on your system. For Windows users, libusb 0.1 is provided through libusb-win32 package. Check the libusb website for updates (http://www.libusb.org).

But how and where do we install it?  libusb-win32 is just a zip file.  Lots of people having trouble with this:

- [Pyusb on windows - no backend available](http://stackoverflow.com/q/13773132/125507)
- [Pyusb on windows 8.1 - no backend available - how to install libusb?](http://stackoverflow.com/q/33972145/125507)
- [libusb installed- but pyUSB backend not found](http://stackoverflow.com/q/11223398/125507)
- [PyUSB ValueError: No backend available](http://stackoverflow.com/q/25756032/125507)
- [PyUSB backend not accessible](http://stackoverflow.com/q/5152133/125507)

michaelllh, 2016-01-12:
I install it on openwrt system.A linux system.I will try to update my libusb.Thank you.

发自网易邮箱大师
在2016年01月12日 23:11，endolith 写道:

The documentation is not very clear on this.

Remember that you need libusb (1.0 or 0.1) or OpenUSB running on your system. For Windows users, libusb 0.1 is provided through libusb-win32 package. Check the libusb website for updates (http://www.libusb.org).

But how and where do we install it? libusb-win32 is just a zip file. Lots of people having trouble with this:

Pyusb on windows - no backend available
Pyusb on windows 8.1 - no backend available - how to install libusb?
libusb installed- but pyUSB backend not found
PyUSB ValueError: No backend available
PyUSB backend not accessible

gtatyous93, 2016-01-13:
all three backends are not working properly. How do you fix this error?

>>> usb.core.find()
2016-01-12 22:02:52,362 ERROR:usb.libloader:Libusb 1 (/usr/local/lib/libusb-1.0.dylib) could not be loaded
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/usb/libloader.py", line 116, in load_library
    return ctypes.CDLL(lib)
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/ctypes/__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/usr/local/lib/libusb-1.0.dylib, 6): no suitable image found.  Did find:
	/usr/local/lib/libusb-1.0.dylib: mach-o, but wrong architecture
2016-01-12 22:02:52,390 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
dlerror: dlopen(/Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so, 1): Symbol not found: _usbi_devices
  Referenced from: /Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so
  Expected in: flat namespace
 in /Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so
fail to load /Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so
2016-01-12 22:02:52,393 ERROR:usb.backend.openusb:Error loading OpenUSB backend
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/usb/backend/openusb.py", line 738, in get_backend
    _ctx = _Context()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/usb/backend/openusb.py", line 512, in __init__
    _check(_lib.openusb_init(0, byref(self.handle)))
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/usb/backend/openusb.py", line 506, in _check
    raise USBError(_lib.openusb_strerror(ret), ret, _openusb_errno[ret])
usb.core.USBError: [Errno None] b'Unspecified kernel/driver failure'
2016-01-12 22:02:52,394 ERROR:usb.libloader:'Libusb 0' could not be found
2016-01-12 22:02:52,394 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/usb/core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError: No backend available

gtatyous93, 2016-01-13:
>>> import usb.core
>>> from usb.backend import libusb1
>>> be = libusb1()

TypeError: 'module' object is not callable

your FAQ has errors!

walac, 2016-01-13:
> TypeError: 'module' object is not callable

> your FAQ has errors!

Fixed, thanks.

walac, 2016-01-13:
> all three backends are not working properly. How do you fix this error?

Your libusb install has problems. I suggest you reach libusb guys out.

gtatyous93, 2016-01-13:
I did contact them and posted my question on stack overflow but I haven't heard from anyone. I installed libusb1 using Homebrew on my Mac OS X, however, I am getting a dlerror because the linker cannot find the backend library. Do you have any ideas on how to fix that? How did you make your backend work properly? 

>>> import usb.core
>>> from usb.backend import libusb1
>>> b = libusb1.get_backend()
>>> b
>>> print(b)
None
>>> dev= usb.core.find()
dlerror: dlopen(/Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so, 1): Symbol not found: _usbi_devices
  Referenced from: /Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so
  Expected in: flat namespace
 in /Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so
fail to load /Library/Frameworks/openusb.framework/Versions/1.1.11/lib/openusb_backend/darwin.so
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/usb/core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError: No backend available
>>> 

Thank you for your time

walac, 2016-01-15:
> I did contact them and posted my question on stack overflow but I haven't heard from anyone. I installed libusb1 using Homebrew on my Mac OS X, however, I am getting a dlerror because the linker [...]

How did you install libusb? Can any program linked to it work?

swinman, 2016-02-08:
FYI for OSX 10.11:
I had previously installed pyusb using `sudo pip3 install pyusb` then subsequently installed libusb1 from source and pyusb was not finding my libusb1.  I then removed the pip pyusb `sudo pip3 uninstall pyusb` and installed pyusb from source
    python3 setup.py build
    sudo python3 setup.py install

and it finds the libusb library just fine..

kvoutyras, 2017-08-02:
@walac: I ran the debug code and it returned:
`2017-08-02` 09:55:22,224 ERROR:usb.libloader:'Libusb 1' could not be found'
`2017-08-02 09:55:22,225 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend'
`2017-08-02 09:55:22,227 ERROR:usb.libloader:'OpenUSB library' could not be found'
`2017-08-02 09:55:22,227 ERROR:usb.backend.openusb:Error loading OpenUSB backend'
`2017-08-02 09:55:22,229 ERROR:usb.libloader:'Libusb 0' could not be found'
`2017-08-02 09:55:22,229 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend'

I tried to install libusb, pyusb separately but didn't work. 

I am running Windows 10-64 bit...

walac, 2017-08-02:
@kvoutyras Did install you libusb through [zadig](http://zadig.akeo.ie/)?

kvoutyras, 2017-08-02:
@walac No I didn't. Unfortunately I don't have admin rights in this pc. Any alternatives?

walac, 2017-08-02:
@kvoutyras I am not sure if you would be able to install libusb successfully without admin rights... Anyway, whats the path of your libusb dll?

kvoutyras, 2017-08-02:
@walac There was no libusb.dll at all. I downloaded it from here: http://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.20/libusb-1.0.20.7z/download

Is there any other place (except for System32 or relevant) I should include this and point the package there?

walac, 2017-08-02:
@kvoutyras when you extract the compressed file the dll is available inside the MS64 or MS32 subfolder (according if you are running with a 64 or 32 bit Windows).

kvoutyras, 2017-08-02:
@walac yes, this is true. SInce I cannot place it at the System32 folder, I could place it only in (e.g.) C:\ProgramData\Anaconda3\Lib\site-packages\usb\backend. Where should I include the path to the .dll library? Thanks! :)

walac, 2017-08-02:
If you put in the same directory of python interpreter, it will work. In general, you can put in any directory that is listed in the `%PATH%` env var, I just don't remember if `cutils` will look at those.

kvoutyras, 2017-08-03:
I put it in `%PATH%` (in my case `C:\ProgramData\Anaconda3\Library\bin`) and worked fine. Thanks!!!

John-K, 2017-08-07:
On Mac OS, lack of permissions may cause this issue. A dangerous way to see if this is what's happening is to run your test script with sudo.

KM4YRI, 2017-08-13:
I ran into the same ```No backend available``` exception when trying to follow the tutorial on a Windows 7 machine.  The following worked for me, per one of the StackOverflow replies:

1. Download the latest Windows binary: https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.21/libusb-1.0.21.7z/download
2. Unzip using 7zip into a temp dir
3. If on 64-bit Windows, copy ```MS64\dll\libusb-1.0.dll``` into ```C:\windows\system32```.  If on 32-bit windows, copy ```MS32\dll\libusb-1.0.dll``` into ```C:\windows\SysWOW64```.

Can I submit a pull request with this info?  Would it be accepted?

walac, 2017-08-13:
> Can I submit a pull request with this info? Would it be accepted?

This has to do with libusb installation. Are you going to submit a PR to add this to readme?

KM4YRI, 2017-08-13:
> This has to do with libusb installation. Are you going to submit a PR to add this to readme?

Yep, I'm asking if you'll accept a pull request on the [README.rst](https://github.com/walac/pyusb/blob/master/README.rst#installing-pyusb-on-windows), with the instructions above appended to the ```Installing PyUSB on Windows``` section.  The current instructions in that section do not provide enough information, as the dozens of comments above have pointed out.

walac, 2017-08-15:
@KM4YRI Yes, sure, open the PR and we can discuss details on the PR.

dustinrwh, 2017-08-25:
All I had to do on Mac was install libusb via homebrew

`brew install libusb`

jcmcken, 2018-02-27:
To anyone having this issue, the solution is that ``libusb-1.0.dll`` needs to be in a directory on your ``PATH`` and readable. Not ``sys.path`` in the Python environment, but ``PATH`` in the process environment. This is where ``ctypes`` looks for DLLs (strangely enough). (You can verify what your path is by running ``print os.environ["PATH"]`` in your Python environment -- or Py3 equivalent)

On many default Windows installations, this is still not enough. Windows can and will prevent you from accessing files placed into e.g. ``C:\Windows\System32`` that it doesn't know about (to prevent things like DLL injection). Therefore the best solution is to do something like create a directory ``DLLs`` in your ``virtualenv`` dir (e.g. ``path/to/your/project/venv/DLLs``), stick the DLL in there, and update your ``PATH`` to include this directory (I would advise against updating the global path, and instead set your interpreter path in the IDE, but that's up to you).

Another thing to keep in mind is whether your Python interpreter is compiled 32-bit or 64-bit. If you try to load the 64-bit DLL in a 32-bit Python environment, it will fail. You can Google around to figure out how to determine one way or the other, or just try both DLLs.

cellularmitosis, 2018-04-29:
Wow, this libusb situation is a complete train wreck for windows users.

Is there any way that we can have pip install libusb automatically when we run pip install pyusb?

cellularmitosis, 2018-04-29:
Ok, I must be missing something fundamental here.

libusb is in system32:

```
c:\Windows\System32>dir libusb-1.0.dll
 Volume in drive C has no label.
 Volume Serial Number is D82F-9527

 Directory of c:\Windows\System32

03/24/2018  08:00 PM           121,856 libusb-1.0.dll
               1 File(s)        121,856 bytes
               0 Dir(s)  124,331,134,976 bytes free

c:\Windows\System32>
```

But python can't see it:

```
import os
print("System32 exists: %s" % os.path.exists("C:\\Windows\System32"))
print("libusb-1.0.dll exists: %s" % os.path.exists("C:\\Windows\System32\\libusb-1.0.dll"))
```

```
System32 exists: True
libusb-1.0.dll exists: False
```

Am I losing my mind?

cellularmitosis, 2018-04-29:
Ok, so it turns out that there's some sort of filesystem redirection happening.  I dropped a copy of libusb-1.0.dll into syswow64 and suddenly python can see the copy in system32:

```
import os
print("System32 exists: %s" % os.path.exists("C:\\Windows\\System32"))
print("libusb-1.0.dll exists: %s" % os.path.exists("C:\\Windows\\System32\\libusb-1.0.dll"))
```

```
System32 exists: True
libusb-1.0.dll exists: True
```

It appears that ctypes can see it now as well:

```
from ctypes.util import find_library
print("find_library: %s" % find_library("libusb-1.0"))
```

```
find_library: C:\Windows\system32\libusb-1.0.dll
```

Unfortunately pyusb still can't see it.

```
import usb.core
usb.core.find()
```

```
Traceback (most recent call last):
  File "example.py", line 28, in <module>
    usb.core.find()
  File "C:\Users\User\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\usb\core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

jcmcken, 2018-04-30:
@cellularmitosis Did you read my previous comment? Try placing the DLL in a non-privileged location and adding that to your ``PATH``

pepasflo, 2018-04-30:
@jcmcken Thanks, I had tried putting it in one of the directories within my user's directory, which was already part of my PATH (I think VS Code added a directory to the Path, and I tried placing it there).  I was surprised that didn't work.

mcuee, 2018-09-19:
Not a pyusb issue. Close this for now. I will see if I can update the Wiki to have better support for Windows users.

jo3lmill3r, 2019-01-16:
Pyusb fixed on OSX High Sierra after 'no backend available issue' -->
Use pip to install packages 'libusb', 'libusb1', and 'pyusb'.
If the library file /usr/local/lib/libusb-1.0.0.dylib doesn't exist then use homebrew to install it.
** After testing with the python script:
import usb.core
dev = usb.core.find()
print(dev)

I got a 'no backend available issue'!!! The usb1 module was expecting to use a library
called 'libusb-1.0.dylib' and not the 'libusb-1.0.0.dylib' that was available.
After making a copy a named 'libusb-1.0.dylib' in the path /usr/local/lib/ 
pyusb worked just fine.

john122984, 2019-02-22:
In Windows (I'm using Windows 10), all I had to do was extract the libusb zip, pick out libusb-1.0.dll, and copy it into the system32 folder.

linxucc, 2019-03-08:
For Mac OS users, just use `brew install libusb`, then the problem solved. 
**Since 'pyusb' rely on 'libusb', for Mac 'libusb' doesn't come as a default. And `pip install pyusb` doesn't install 'libusb' automatically. In 'pyusb' something like 'libusb','openusb' is called a 'backend'.  So you have to manually install it.**
Hope this info helps. 

**And to developers:** I suggest that some additional help information be prompt when this 'no backend exception' occurs, something like 'maybe libusb or openusb is not installed yet? ' would help a lot.

MrSassyBritches, 2020-01-09:
Dropping the dll into my system32 folder did not work, and moving it to a non-privileged folder and added that folder to the path, still cannot import `usb.core`. At the same time, I copied cellularmitosis 3 line script and produced the following output. So libusb is present, but not able to be imported.

```
System32 exists: True
libusb-1.0.dll exists: True
find_library: C:\WINDOWS\system32\libusb-1.0.dll
```

jonasmalacofilho, 2020-01-10:
@MrSassyBritches 

> Dropping the dll into my system32 folder did not work, and moving it to a non-privileged folder and added that folder to the path, still cannot import `usb.core`. At the same time, I copied [...]

For me it seems that your issue is not with libusb (as `find_library` finds it just fine), but with PyUSB itself.  How did you install it, and what version do you have?  Any chance you installed to a different (virtual) environment?

jonasmalacofilho, 2020-01-12:
An error when *importing* `usb.core` suggests that this is not the same problem (i.e. that the exception is *not* `ValueError('No backend available')`).  The ability of `find_library` to locate the libusb DLL further reinforces that.

Based on this, I'm closing this to avoid more spam to everyone already involved in this (old) thread.  Even if Tollis' problem has some relation to a "no backend error", **let's please start a new thread;** it would likely be caused by something different anyway.

---

@MrSassyBritches

Besides the questions in the previous comment, can you also confirm the exception/error you see?  Please include any tracebacks as well (in a new issue).

closed 2020-01-12 by jonasmalacofilho

MrSassyBritches, 2020-01-20:
Sorry I didn't respond! It looks like installing pyusb globally with pip fixed the issue. I don't have the exact info but using pipenv, an older version was installed. 

If I see the issue again, because I will use pipenv again, I will open a new issue.

amendoncabh, 2020-08-24:
PyUSB over windows 10 using libusb-1.2 DLL´s.

I did use the "libusb-win32-bin-README.txt" reference file downloaded [here](https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/), to install files on correct windows folder by my system version.
In special aptention to 32bits files (odoo 10), because you will need put files on syswow64 windows folder.

And "voilá" solved problem!

jonasmalacofilho, 2020-08-24:
libusb-win32 is a port of libusb-0.1; their 1.2.6 version number is confusing and has nothing to do with libusb-1.0.

These days is better to use libusb-1.0.  You can find the latest releases in the [libusb/libusb](https://github.com/libusb/libusb/releases) repository; the DLLs are in the `*.7z` files.

---

> Based on this, I'm closing this to avoid more spam to everyone already involved in this (old) thread.  
> [...] **let's please start a new thread** [...]