pyusb/pyusb #123: Release 1.0.0rc1 on PyPI?

padelt, 2016-01-07:
Hi there,
Over at https://github.com/padelt/temper-python/ we are depending on `pyusb>=1.0.0rc1` to get all our target platforms to work. 
On local installations, we get away with having a `dependency_links` setting in the [PyPI package metadata](https://github.com/padelt/temper-python/blob/master/setup.py#L15).
Since we published that to [PyPI](https://pypi.python.org/pypi/temperusb) we see that `pip install temperusb` fails. Obviously the git version is not pulled in but only the published versions in PyPI are considered. You can see the effect [here](https://github.com/padelt/temper-python/issues/41#issuecomment-169626148).

Anyway, Would it be possible to get a release of current master to PyPI as 1.0.0rc1? That would hopefully help our users get an easily installable version to read out their Temper USB thermometers.
-- Philipp

walac, 2016-01-07:
I am still catching up from a long term health problem, and I dedicating most of my free time for recovery. I am releasing rc1 as soon as I can.

closed 2016-01-07 by walac

padelt, 2016-01-07:
Thank you and get well soon!

ps-jay, 2016-04-15:
Thanks for releasing it!