pyusb/pyusb #126: Different behaviour from ctrl_transfer with linux and windows backend

khillman, 2016-01-22:
Hi, i'm not sure whether this is a problem regarding pyusb itself or libusb or a wrong usage of the library...

If i issue something like this in linux:

```python
device = usb.core.find(idVendor=vendorVal, idProduct=productVal)
if device:
    if device.is_kernel_driver_active(0):
        try:
            device.detach_kernel_driver(0)
            print "kernel driver detached"
        except usb.core.USBError as e:
            raise Exception("kernel driver in use - detaching failed with %s" % e
    device.set_configuration()
    device.ctrl_transfer(bmRequestType=0x21, bRequest=9, wValue=0x0200, wIndex=0,
                data_or_wLength = data)
```

Everything is working as expected - data is being sent to the device using ctrl_transfer as the device has no OUT endpoint.
Doing the same in windows tells me first that is_kernel_driver_active and detach_driver_active is not implemented after removing that the ctrl_transfer fails with an timeout error even if i set a really high timeout value...

Any ideas how to fix that and send messages to an usb device without OUT endpoint?

walac, 2016-01-25:
The kernel driver functions only make sense in *nix, because no drivers must be attached to the device. In Windows it is quite the opposite, as a supported driver must be loaded to communicate with the device, so these functions have no role. The best way is to call them and ignore any `NotImplementedError` exception.

khillman, 2016-01-26:
Well but if they does not have/play a role (because they are not implemented and therefore no-ops) the problem must be something else or not? I'll test whether calling them and catching the errors will work but i don't see a high probability that this could work. I've used the libusb backend from http://libusb.info/ in version 1.0.20 but every time i tried i got an timeout exception - should i put the stack trace to the post? I've also analyzed the usb traffic via usbmon and saw that there were no URB OUT traffic as it should be and is the case in linux.

walac, 2016-01-26:
This might be a bug in libusb then, I suppose. Could you please use `PYUSB_DEBUG=debug` to generate logs for both Linux and Windows to make sure the PyUSB behavior is, in general, the same in both systems?

khillman, 2016-02-02:
Sorry for the delay - i was not able to generate the logs earlier. 

The following test code:

```python
import usb.core
device = usb.core.find(idVendor=idVendor, idProduct=idProduct)
if device:
    try:
        if device.is_kernel_driver_active(0):
            device.detach_kernel_driver(0)
            print "kernel driver active"
    except NotImplementedError as e:
        pass
    device.set_configuration()
    device.ctrl_transfer(bmRequestType=0x21, bRequest=9, wValue=0x02200, wIndex=0, data_or_wLength='\x02\x1bP')
```
leads in linux to:

```
sudo python test-script.py 
2016-02-02 08:24:29,506 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 1ab4e00 at 7faee693bc50>)
2016-02-02 08:24:29,509 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2016-02-02 08:24:29,509 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2016-02-02 08:24:29,509 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee6953410>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee6953550>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee6953450>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee69534d0>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee6953410>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee6953550>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee6953450>)
2016-02-02 08:24:29,510 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7faee69534d0>)
2016-02-02 08:24:29,511 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x7faee69534d0>)
2016-02-02 08:24:29,511 DEBUG:usb.backend.libusb1:_LibUSB.is_kernel_driver_active(<usb.backend.libusb1._DeviceHandle object at 0x7faee8ef6910>, 0)
2016-02-02 08:24:29,511 DEBUG:usb.backend.libusb1:_LibUSB.detach_kernel_driver(<usb.backend.libusb1._DeviceHandle object at 0x7faee8ef6910>, 0)
kernel driver active
2016-02-02 08:24:29,511 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7faee69534d0>, 0)
2016-02-02 08:24:29,511 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x7faee8ef6910>, 1)
2016-02-02 08:24:29,511 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x7faee8ef6910>, 33, 9, 8704, 0, array('B', [2, 27, 80]), 1000)
2016-02-02 08:24:29,515 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x7faee8ef6910>)
2016-02-02 08:24:29,515 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
```

whereas in windows it leads to:

```
python test-script.py
2016-02-02 08:28:14,496 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\Python27\libusb-1.0.dll', handle 6b800000 at 2f6b130>)
2016-02-02 08:28:14,496 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2016-02-02 08:28:14,496 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2016-02-02 08:28:14,529 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x02FC5930>)
2016-02-02 08:28:14,529 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x02FC5A50>)
2016-02-02 08:28:14,529 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x02FC5A10>)
2016-02-02 08:28:14,529 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x02FC5A10>)
2016-02-02 08:28:14,543 DEBUG:usb.backend.libusb1:_LibUSB.is_kernel_driver_active(<usb.backend.libusb1._DeviceHandle object at 0x02F2D9D0>, 0)
2016-02-02 08:28:14,543 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x02FC5A10>, 0)
2016-02-02 08:28:14,543 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x02F2D9D0>, 1)
2016-02-02 08:28:14,543 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x02F2D9D0>, 0)
2016-02-02 08:28:14,543 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x02F2D9D0>, 33, 9, 8704, 0, array('B', [2, 27, 80]), 1000)
Traceback (most recent call last):
  File "test-script.py", line 15, in <module>
    device.ctrl_transfer(bmRequestType=0x21, bRequest=9, wValue=0x02200, wIndex=0, data_or_wLength='\x02\x1bP')
  File "C:\Python27\lib\site-packages\usb\core.py", line 971, in ctrl_transfer
    self.__get_timeout(timeout))
  File "C:\Python27\lib\site-packages\usb\_debug.py", line 60, in do_trace
    return f(*args, **named_args)
  File "C:\Python27\lib\site-packages\usb\backend\libusb1.py", line 819, in ctrl_transfer
    timeout))
  File "C:\Python27\lib\site-packages\usb\backend\libusb1.py", line 552, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 22] Invalid parameter
2016-02-02 08:28:14,591 DEBUG:usb.backend.libusb1:_LibUSB.release_interface(<usb.backend.libusb1._DeviceHandle object at 0x02F2D9D0>, 0)
2016-02-02 08:28:14,591 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x02F2D9D0>)
2016-02-02 08:28:14,607 DEBUG:usb.backend.libusb1:_LibUSB.__del__()
```

Any ideas?

khillman, 2016-02-03:
I've found at least a work around for the problem - the problem lies in the data parameter:
```python
data_or_wLength='\x02\x1bP'
```

In linux libusb seems to automatically add some padding to the data so the needed size for the device is reached (in my case 64 bytes). In contrast in windows the exactly same call seems to result in an input/output error (the above invalid parameter error is because wValue has to be 0x0200). 

The workaround so far for me is to give an byte array of the given length as input to data_or_wLength and therefore make the data size more explicit.
So the corrected ctrl_transfer line would be that:

```python
    device.ctrl_transfer(bmRequestType=0x21, bRequest=9, wValue=0x0200, wIndex=0, data_or_wLength=array.array('B', [0x02, 0x1b, 0x50, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]))
```

Which is working within linux AND windows ;)

closed 2016-02-03 by khillman