pyusb/pyusb #128: Have a field to tell me the currently used backend.

benallard, 2016-01-30:
I remember we talked about this once, but I can't remember the outcome of the situation.

I'd love to be able to add this information to the bug report my users are sending, to be able to tell them "switch to libusb1" if they're not using it.

Any way to get this implemented ?

walac, 2016-01-30:
I could swear I implemented a `backend` property to return this information.

benallard, 2016-02-01:
Beautiful, thanks !

benallard, 2016-02-01:
One small question though, does it has to be on a usb device ? Can't I have this as a property of the module ? It would make it way easier to access ! Or is there any way to create a dummy device in order to read that property ?

walac, 2016-02-03:
Unfortunately no, backend is tied to device, not module.

benallard, 2016-02-03:
That's pretty unfortunate. Would you consider at least adding the backend name to the ``USBError`` Exception ? That's almost all what I have left when It's time to consider switching to another backend ...

Or what's your recommendation on this ? Always recommend libusb1 when something else is detected ?

My main issue are all the 'Device busy' that could be avoided if we could detach the kernel driver ...

walac, 2016-02-18:
I can add it, but only after stable release.

jonasmalacofilho, 2019-12-10:
Hi @benallard,

I recently became a maintainer here and I'm checking all the old issues...

_If possible,_ I would prefer to find a solution to the underlying issue – deciding whether the end-user needs to switch to the libusb1 backend – that doesn't add the backend object/name to `USBError`.

There is a lot of information that can be useful to have in the exception, depending on the use case, and the backend seems to only cover a very narrow subset of scenarios.

Does the real world problem by any chance have to do with galileo, specifically [galileo/#94](https://bitbucket.org/benallard/galileo/issues/94/usbcoreusberror-errno-16-resource-busy) and its duplicates?

closed 2020-01-05 by jonasmalacofilho