pyusb/pyusb #130: no backend available on BusyBox 1.16

TonyEckel, 2016-02-18:
Hi - Hope you are well.

First - Newbie with Python and have read ALL the "no backend available" postings I could find. 

Running on a Synolgy BusyBox v1.16.1 (Linux 2.6.32.12) as root ('cause I can)

Trying to get the pyusb running so I can mess with the APC UPS and monitor power .... 

Been struggling with the simple stuff and I am pretty stumped at this point: Debug below.

```
BusyBox v1.16.1 (2015-11-12 18:01:35 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

XXXXXX-NAS> python3
Python 3.4.1 (default, Oct 31 2014, 16:38:27)
[GCC 4.3.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path
['', '/volume1/@appstore/py3k/usr/local/lib/python3.4/site-packages/setuptools-20.1.1-py3.4.egg', '/volume1/@appstore/py3k/usr/local/lib/python3.4/site-packages/libusb1-1.4.1-py3.4.egg', '/volume1/@appstore/py3k/usr/local/lib/python34.zip', '/volume1/@appstore/py3k/usr/local/lib/python3.4', '/volume1/@appstore/py3k/usr/local/lib/python3.4/plat-linux', '/volume1/@appstore/py3k/usr/local/lib/python3.4/lib-dynload', '/volume1/@appstore/py3k/usr/local/lib/python3.4/site-packages']
>>> import os
>>> os.environ['PYUSB_DEBUG'] = 'debug'
>>> import usb.core
>>> usb.core.find()
2016-02-17 19:33:57,496 ERROR:usb.libloader:'Libusb 1' could not be found
2016-02-17 19:33:57,498 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2016-02-17 19:33:57,509 ERROR:usb.libloader:'OpenUSB library' could not be found
2016-02-17 19:33:57,510 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2016-02-17 19:33:57,540 ERROR:usb.libloader:'Libusb 0' could not be found
2016-02-17 19:33:57,540 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/volume1/@appstore/py3k/usr/local/lib/python3.4/site-packages/usb/core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError: No backend available
>>>

```

Installed libusb with pip, easy-install and manually: tried them all.

Looking for troubleshooting guidence.

Thanks

walac, 2016-02-18:
It feels like something wrong while loading libusb0 library. How did install it?

TonyEckel, 2016-02-22:
To be clear - libusb0 is also required? - it was not clear (to me)

Thanks

walac, 2016-02-25:
Either libusb 0.1 or libusb 1.0 is required, but not necessarily both

mcuee, 2018-08-07:
Closing old issues. The answer is already quite clear.

closed 2018-08-07 by mcuee