pyusb/pyusb #132: libusb0 backend does not implement usb.backend.IBackend completely

SimplicityGuy, 2016-02-21:
`libusb0` backed does not currently implement the `usb.backend.IBackend` entirely. Specifically it's missing an implementation of `attach_kernel_driver` and `is_kernel_driver_active`.

SimplicityGuy, 2016-02-21:
In some investigation here, it may not be possible to implement `attach_kernel_driver`.

However, `is_kernel_driver_active` may be implemented by using this in `_setup_prototypes`:
```
    # linux only

    # int usb_get_driver_np(usb_dev_handle *dev,
    #                       int interface,
    #                       char *name,
    #                       unsigned int namelen);
    if hasattr(lib, 'usb_get_driver_np'):
        lib.usb_get_driver_np.argtypes = [_usb_dev_handle,
                                          c_int,
                                          c_char_p,
                                          c_int]
```

walac, 2016-02-21:
Hrm, good spot, I will fix this as stable release.

commit 62bddae8 (2019-12-10): Implement is_kernel_driver_active on the libusb0 backend

Fixes: #132 ("libusb0 backend does not implement usb.backend.IBackend
completely")
Reported-by: Robert Wlodarczyk (@SimplicityGuy)
closed 2019-12-10 by jonasmalacofilho