pyusb/pyusb #136: i did as u said To debug what's wrong and ran that debug code and got the following errors:

dineshch009, 2016-03-17:
2016-03-17 12:33:24,976 ERROR:usb.libloader:Libusb 1 (c:\Python27\libusb-1.0.dll) could not be loaded
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\usb\libloader.py", line 114, in load_library
    return lib_cls(lib)
  File "C:\Python27\lib\ctypes\__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 193] %1 is not a valid Win32 application
2016-03-17 12:33:24,990 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2016-03-17 12:33:24,990 ERROR:usb.libloader:'OpenUSB library' could not be found
2016-03-17 12:33:24,990 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2016-03-17 12:33:24,990 INFO:usb.core:find(): using backend "usb.backend.libusb0"
2016-03-17 12:33:24,990 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
2016-03-17 12:33:24,990 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x01FC6580>)

ixjlyons, 2016-03-17:
Could you reference or just state what debugging procedure you're referring to?

Anyway, I've seen this error before (`%1 is not a valid Win32 application`), and in my case it indicated a mismatch between Python and the DLL being loaded (one was 64-bit and the other was 32-bit). It might be helpful to know if your installed Python and libusb-1.0 are 32-bit or 64-bit. My guess is you have 64-bit Python and a 32-bit library.

dineshch009, 2016-03-18:
Thanks for the reply i have sorted out the problem. I used 32-bit Python
and 32-bit library but the problem is that i didn't have the libusb-1.0.dll
and libusb0.dll. Which i have downloaded them and included them in PATH.
Now everything is working fine and my code is deteting every USB devices
except the one device which i want :(

closed 2016-04-19 by walac

dineshch009, 2019-08-04:
Hey,

I just signed the petition "Central Government of India: Indian
healthsystem at stake. Save yourselves. Say no to NMC" and wanted to see if
you could help by adding your name.

Our goal is to reach 10,000 signatures and we need more support. You can
read more and sign the petition here:

http://chng.it/wpbdXJDW9F

Thanks!
Dinesh