pyusb/pyusb #139: ValueError: The device has no langid

ddurs555, 2016-04-14:
Hopefully I am not posting in the wrong format here, however I have a project I have been working on for some time now that is built on pyusb 1.0rc, raspberry pi etc..

I am connecting around 90+ of the same USB devices and usually on the first attempt of the sequence.

devices = usb.core.find(idProduct... idVendor.... find_all=True)
for dev in devices:
    devices.set_configuration()
    print device.serial_number

pyusb will throw ValueError: The device has no langid on N number of devices (Some report properly some don't)

However if I try this same sequence of code again, it works flawlessly.

1. What might be the underlying issue?
2. If the underlying source is unknown do you think adding an attempt loop here:
https://github.com/walac/pyusb/blob/master/usb/util.py#L311
would be prudent?

walac, 2016-04-19:
This looks like a bug on the device side.

tormodvolden, 2016-06-17:
Yup. Does lsusb -v always print the serial number correctly? Is this the same kind of devices as in bug #145?

ivanelson, 2016-08-19:
I also get this message to a USB printer. I noticed that on the same Linux command "lsusb -v" only displays this printer if the user is root.

It seems that some USB devices require the command to run as root.

`dev = usb.core.find(product='FS800i', idVendor=0x23b8,idProduct=0x0003)`

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1269, in find
    return _interop._next(device_iter(**args))
  File "/usr/local/lib/python2.7/dist-packages/usb/_interop.py", line 66, in _next
    return next(iter)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1249, in device_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1248, in <genexpr>
    tests = (val == getattr(d, key) for key, val in kwargs.items())
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 841, in product
    self._product = util.get_string(self, self.iProduct)
  File "/usr/local/lib/python2.7/dist-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid

But as root it works:
`dev = usb.core.find(product='FS800i', idVendor=0x23b8,idProduct=0x0003)`
`
I do not know if this behavior is weird ?!

ddurs555, 2016-08-19:
> File "", line 1, in

I solved my problem by forcing the LANG_ID
Something like this might work for you:

device._langids = (1033,)

ivanelson, 2016-08-22:
ddurs555,
Where you entered this code?
`device._langids = (1033,)`

ddurs555, 2016-08-22:
Please note that I do run this code as root and that your core issue simply might be a permissions issue.  Mine however is a bad device that doesn't adhere to the proper protocol.

                devs = usb.core.find(idVendor=device_type[0],idProduct=device_type[1], find_all=True)
                for device in devs:
                    try:
                        device.set_configuration()
                    except usb.core.USBError as usb_error:
                        if 'Resource busy' in str(usb_error):
                            if device.is_kernel_driver_active(0):
                                device.detach_kernel_driver(0)
                                device.set_configuration()
                    device._langids = (1033,)
                    devices.append(device)

walac, 2016-08-24:
Notice `get_string` accepts the langid parameter, and then it will use this value instead of trying to retrieve it from the device.

ivanelson, 2016-08-24:
This problem in "langid" only occurs in "USB" devices ACM type. For example Ubuntu, ttyACM0 ... ttyACM...N
To work around the problem you must pass the "Vendor" and "Product" in the signature of the method: usb.core.find (idVendor = device [1], idProduct = device [2]).

And also create rules for the device:

cat /etc/udev/rules.d/70-persistent-usb.rules 
SUBSYSTEMS=="usb", ATTRS{idVendor}=="23b8", ATTRS{idProduct}=="0005", ```
ATTRS{manufacturer}=="DARUMA TECNOLOGIA", SYMLINK+="ttyACM5", MODE=="0777"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="23b8", ATTRS{idProduct}=="0003", ATTRS{manufacturer}=="Daruma Telecomunicacoes e Informatica S/A", SYMLINK+="ttyACM5", MODE=="0777"
```

tormodvolden, 2016-08-24:
Sorry Ivanelson, but that is not a good example of udev rules to create. First, 70-persistent-usb.rules is an automatically generated file so you shouldn't change it - instead add your rules to a new file with a filename that you make up yourself. If you match on idVendor you most likely don't have to match on manufacturer as well. Creating a symlink that potentially will conflict with another device is bad - the autogenerated file will do things like this though, to make devices names persistent. And finally, whereas it is convenient to throw file mode 777 at it, it is not an example to follow. Better assign the device to a group, and make sure the user is member of it.

Also this is not relevant to the original poster's issue, he's running as root. However in some of the other comments it is a question of permissions: If you don't have permissions you cannot retrieve the langid nor the serial number/manufacturer/product strings. The langid is retrieved first in all cases, so that's the error that get displayed when the user tries to get e.g. the serial number.

kdeme, 2016-08-31:
I was trying to control a Keysight InfiniiVision MSO with PyVISA + PyVISA-py backend (using pyusb) and came across this same error. It was not an permission issue as it also happens when root.

Initially I tested by just setting the langid in usb/util.py to 0 in case 0 == len(langids). That worked, but is not a real solution obviously.

Now I tested the code of @ddurs555 and this solves the issue also. Much cleaner as workaround. In case of my device I can actually set the langid to whatever (even 0 although it is not even a valid langid). I guess it does not look at langid at all.

jedie, 2017-04-14:
In https://github.com/rsc-dev/loophole/issues/8 is a similar Problem: The program just raise `ValueError: The device has no langid`

Currently we do this:
```
                try:
                    info = Device.get_info(dev)
                except ValueError as err:
                    print "Device no: %i" % i
                    print "Device info:"
                    print dev
                    print "-"*79
                    if 'langid' in err.message:
                        raise ValueError(
                            (
                                "Can't get device info. Origin Error: %s\n"
                                "Maybe this is a permission issue.\n"
                                "Please read section 'permission' in README ;)"
                            ) % err
                        )
                    raise # raise origin error
```

Question: Is it possible to explicit check if the user has the right permissions ?!?

walac, 2017-04-17:
> Question: Is it possible to explicit check if the user has the right permissions ?!?
No, as this is system specific and encapsulated inside libusb.

coutliuxing, 2017-05-09:
change like this >>> `usb.util.get_string(dev, dev.iSerialNumber)`

jonasmalacofilho, 2019-12-12:
There are a few issues combined here...

---

**Permission error**

Reading a string descriptor requires permission to talk to the device, which by default is reserved to root on most Linux distros.

Those experiencing this should run their programs with root privileges or, alternatively, expose the devices to their user/group with udev rules.

---

**Devices that are not compliant to the USB spec**

Devices should provide a LANGID array if they provide any string descriptors, but unfortunately it seems that not all do.

I've look into this in a lot of detail and there's nothing we can safely do.  If the developer knows that a particular offending device secretly supports a given LANGID (perhaps even a reserved/ilegal LANGID like `0x0000`), then they can manually specify it:

```python
# instead of
foo = dev.serial_number

# try writing
langid = 0x???? # langid that device secretly accepts
foo = usb.util.get_string(dev, dev.iSerialNumber, langid=langid)
```

_(requires commit ca2e23bb4f00b51b493c860a5765af97681d4fe7)_

---

**Devices that are compliant but that might not immediately respond after being configured**

Calling `set_configuration()` configures the device.  Presumably some devices may need some time before they are fully operational after that.  This would explain the reports where the issue only manifests every once in a while, particularly only on the first `set_configuration()`.

Those experiencing this should experiment with adding a small delay after `set_configuration()`.  Additionally, it might be useful to remember that the device doesn't have to configured before descriptors are fetched (including string descriptors like `iSerial`).

Lastly, `set_configuration()` can also cause a soft-reset if the device was already in the desired configuration, which in turn causes other issues (see [FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration)).

---

**Devices that don't expose any string descriptors**

Per the USB specification, devices that don't expose any string descriptors also wont return a LANGID array.

closed 2019-12-12 by jonasmalacofilho

kobena, 2020-05-12:
Most of these problems for those using linux comes from not setting the udev rule
            ▪ Create text file /etc/udev/rules.d/99-garmin.rules with contents:
            ▪ SUBSYSTEM=="usb", ATTR{idVendor}=="0fcf", ATTR{idProduct}=="1008", MODE="666"
            ▪ Note that the idVendor and idProduct values are in hex already so don’t prefix 0x. If for all usb you should not put anything in there.
            ▪ restart udev and reconnect the usb device.

bahargraviton, 2020-06-08:
> Please note that I do run this code as root and that your core issue simply might be a permissions issue. Mine however is a bad device that doesn't adhere to the proper protocol.
>
> ``` [...]

I am a beginer and I have to connect my instrument which is "Tektronix DPO" to the raspberry pie. I found the code for PyVisa, but when I wanted to run, it has this error. I do not know what is going on. I am also not sure that if I have NI visa for backened. My question is that where do you eneter these codes?

jonasmalacofilho, 2020-06-08:
The most common cause for `ValueError: The device has no langid` is simply lack of permission to access the device.  Try running your program with `sudo` and, if that works, consider using a udev rule to grant access to the device to the unprivileged users.

---

The [other possible causes and partial solution](https://github.com/pyusb/pyusb/issues/139#issuecomment-565114069) will likely require you to change PyVisa code (unless they expose and raw `usb.core.Device` handle for you through some API).

Popolon, 2021-08-05:
If it can help, I had a similar error code due to a bug in suspend mode. It can be resolved/tested by removing the autosuspend for all usbdevices the following way. On some systems the value to echo is <code>0</code> instead of <code>-1</code>:

<pre>echo -1 >/sys/module/usbcore/parameters/autosuspend</pre>

Resolve this permanently after reboot depend on system and devices, refer to your distribution documentation about USB and autosuspend for this.

jonasmalacofilho, 2021-08-06:
> If it can help, I had a similar error code due to a bug in suspend mode [...]

I'm guessing the issue is that your handles were no longer valid after your USB devices power cycled during the suspended period.

Popolon, 2021-08-09:
In my case, it's during flashing,  when autosuspend is not disabled, there is an error with the device and disappear from device list when doing lsusb just after. If autosuspend is disabled, flash works well. An error message appear at verification (re-read the board), don't know why, but the board is correctly and device still here with lsusb.