pyusb/pyusb #142: Was rc1 removed from PyPI when 1.0.0 was released?

estan, 2016-05-06:
Hi there,

We use pinned requirements and had our pyusb requirement pinned to 1.0.0rc1. Our builds recently broke with:

```
Collecting pyusb==1.0.0rc1                                                                                                                                  
  Could not find a version that satisfies the requirement pyusb==1.0.0rc1 (from versions: 1.0.0a2, 1.0.0a3, 1.0.0b1, 1.0.0b2, 1.0.0)
```

Was 1.0.0rc1 removed from PyPI when 1.0.0 was released? If so, why?

walac, 2016-05-06:
I am not sure what happened, 1.0.0rc1 is still there (just checked).

estan, 2016-05-06:
Hm. Yea, I know the page for it on PyPI is still there, but installation through `pip` does not seem to work. This is from my laptop just now:

```
[estan@pyret ~]$ virtualenv2 -p python3 ~/env
Running virtualenv with interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/estan/env/bin/python3
Also creating executable in /home/estan/env/bin/python
Installing setuptools, pip, wheel...done.
[estan@pyret ~]$ source ~/env/bin/activate
(env) [estan@pyret ~]$ pip install pyusb==1.0.0rc1
Collecting pyusb==1.0.0rc1
  Could not find a version that satisfies the requirement pyusb==1.0.0rc1 (from versions: 1.0.0a2, 1.0.0a3, 1.0.0b1, 1.0.0b2, 1.0.0)
No matching distribution found for pyusb==1.0.0rc1
(env) [estan@pyret ~]$
```

Would you be able to test this on your side?

Did the artifact for 1.0.0rc1 somehow get removed from the PyPI package during the upload of 1.0.0 final? (Sorry, I'm not entirely sure how deployment to PyPI works).

estan, 2016-05-06:
The [page at PyPI](https://pypi.python.org/pypi/pyusb/1.0.0rc1) is still there, but it no longer lists any source distributions for it. The source distribution for [1.0.0b2](https://pypi.python.org/pypi/pyusb/1.0.0b2) is still there (and 1.0.0a2, 1.0.0a3, 1.0.0b1, and 1.0.0 of course).

walac, 2016-05-08:
I probably removed the package by accident while uploading the stable release. Unfortunately, pypi doesn't allow file re-upload. Is it possible for you to migrate to stable release?

estan, 2016-05-08:
Yes, no problem at all, we'll migrate to 1.0.0. Was just surprised :)

If it's not possible then it's not possible, so closing this now.

closed 2016-05-08 by estan