pyusb/pyusb #143: Access device underlying file descriptor

rnixx, 2016-05-09:
I want to implement a ``USBPort`` for the Twisted main loop similar to the ``SerialPort`` implementation (https://github.com/twisted/twisted/blob/trunk/twisted/internet/_posixserialport.py#L23). Twisted works with poll/select/epoll, depending on the platform, thus I think I need to acquire access to the device underlying file descriptor. Is there a supposed way to do so?

rnixx, 2016-05-10:
Here's a related but unanswered question on stack overflow
http://stackoverflow.com/questions/28790949/integrate-pyusb-device-into-poll-loop
May someone bring enlightenment to this please?

walac, 2016-05-10:
This is related to issue #63. I didn't think much ahead of the design yet, but the general idea is to have async function which will return a `Future` compatible with Python I/O frameworks.

rnixx, 2016-05-10:
Actually Twisted's reactor (main loop) does epoll.register on it's transport objects. I tried with the value returned from ``device.backend.lib.libusb_get_pollfds(device.backend.ctx)`` but just got an IOError "Bad file descriptor". Thus said, for integrating to Twisted in a way it's supposed to be it's required to get the related fd.

Just to get an idea whether my approach has a chance to work - Can you tell me how to access the correct fd?

Would it be an option to add a function fileno() to device/endpoint (wherever it makes sense) objects for cases as mine w/o the need for waiting for a complete async implementation (which I fear would not help me that much anyway in Twisted)

BTW - what do you mean with ``Future``?

Thanks

walac, 2016-05-10:
Currently there is no interface to get the underlining fd from libusb, but it can be implemented. By `Future`, I mean [this](https://docs.python.org/3/library/concurrent.futures.html#concurrent.futures.Future). I never worked with Twister, only with Tornado, which uses the concept of a Future, and Python 3 is also embracing it, that's why this would be my choice.

rnixx, 2016-05-10:
I added a dedicated issue with the feature request here #144

closed 2016-05-10 by rnixx