pyusb/pyusb #144: Feature Request: Expose device underlying file descriptor via API

rnixx, 2016-05-10:
As described in #143, it would be great if devices provide the underlying file descriptor via API.

jonasmalacofilho, 2019-12-08:
Cc: @mcuee 

This would require patches to the backend libraries, as well as be linux[/mac] specific, no?

mcuee, 2019-12-09:
I think libusb api has that.
http://libusb.sourceforge.net/api-1.0/group__poll.html

I think it is generic, Windows libusb implementation has emulated poll-fd as well.
https://github.com/libusb/libusb/tree/master/libusb/os

mcuee, 2019-12-09:
Oops, looks like we do have a limitation under Windows. But it may still be okay, just return unsupported under WIndows (same for hotplug now for libusb Windows).

https://github.com/libusb/libusb/wiki/Windows
Because there is no native poll() on Windows, the ability to return externally pollable file descriptors on Windows libusb_get_pollfd() returns an error.

mcuee, 2021-07-24:
@jonasmalacofilho Is this worth looking at? Thanks.

jonasmalacofilho, 2021-07-24:
I think it is, and IIRC getting the fd has also been mentioned in the Termux/Android discussions.