pyusb/pyusb #145: "Error: The device has no langid"

patrickmk, 2016-05-13:
I am attempting to connect to a Tektronix DPO 2002B oscilloscope using pyusb, and when I attempt to run:
`address = 'USB0::0x0699::0x0399::<serial>::INSTR'

import visa #import pyVISA
rm = visa.ResourceManager('@py')

print rm.list_resources()
#scope = rm.open_resource(address) #open the oscilloscope
#print scope.query('*IDN?')`

I get the error of: 

`Traceback (most recent call last):

  File "<ipython-input-3-8d9e627f2f5d>", line 1, in <module>
    runfile('/home/iago/Documents/Code/test_connection_tektronix.py', wdir='/home/iago/Documents/Code')

  File "/home/iago/anaconda2/lib/python2.7/site-packages/spyderlib/widgets/externalshell/sitecustomize.py", line 699, in runfile
    execfile(filename, namespace)

  File "/home/iago/anaconda2/lib/python2.7/site-packages/spyderlib/widgets/externalshell/sitecustomize.py", line 81, in execfile
    builtins.execfile(filename, *where)

  File "/home/iago/Documents/Code/test_connection_tektronix.py", line 20, in <module>
    print rm.list_resources()

  File "/home/iago/anaconda2/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)

  File "/home/iago/anaconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])

  File "/home/iago/anaconda2/lib/python2.7/site-packages/pyvisa-py/usb.py", line 169, in list_resources
    serial_number=dev.serial_number,

  File "/home/iago/anaconda2/lib/python2.7/site-packages/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)

  File "/home/iago/anaconda2/lib/python2.7/site-packages/usb/util.py", line 315, in get_string
    raise ValueError("The device has no langid")

ValueError: The device has no langid`
 
This same problem arises from attempting to use open_resource(address)
I was able to successfully connect to a LeCroy oscilloscope over TCP/IP, which leads me to believe there is likely an issue with using PyUSB, since in the file "util.py", the value of dev.langids is empty (called in line 311). 

Thank you for your help!

walac, 2016-05-16:
You probably should contact PyVISA developer first.

tormodvolden, 2016-06-17:
Does lsusb -v list the serial number correctly?

mcuee, 2018-09-19:
Close this for now as it is not related to pyusb.

closed 2018-09-19 by mcuee

mcuee, 2021-07-31:
Please refer to #179 and now this issue should be resolved.