pyusb/pyusb #147: Implement Device.__eq__()

martijnthe, 2016-05-25:
I'd like to be able to write something like:

```
devs = set()
devs |= set(usb.core.find(find_all=True, custom_match=matcher_one))
devs |= set(usb.core.find(find_all=True, custom_match=matcher_two))
```

If `matcher_one` and `matcher_two` both match device XYZ, `devs` will contain two instances of `Device` referring to XYZ. I'd expect `devs` to contain only one instance referring to XYZ.

jonasmalacofilho, 2021-01-19:
Thoughts on implementing `__eq__` as follows?

```py
def __eq__(self, other):
    return (self.backend, self.bus, self.address) == \
           (other.backend, other.bus, other.address)
```

Essentially defining equality based on the OS's view of the USB tree, but also taking into account the different possible backends.

Cc: @mcuee

mcuee, 2021-01-20:
Looks good to me.

commit 251dbc15 (2021-02-03): core: implement Device.__eq__

Essentially defines equality based on the OS's view of the USB tree, but
also takes into account the different possible backends.

Closes: #147 ("Implement Device.__eq__()")
closed 2021-02-03 by jonasmalacofilho