pyusb/pyusb #148: regression in legacy module

tormodvolden, 2016-06-17:
In 0.4 and 1.0alpha3 it was possible to pass a list of bytes to bulkWrite(). Now it fails. Unfortunately I never tested this on the betas :(

 buffer:  ['\x07', '\xb8', '\x0b', '\x00', '\x00']

    error = self.handle.bulkWrite(EP, buffer, TIMEOUT)
  File "/home/tormod/pyusb/usb/legacy.py", line 148, in bulkWrite
    return self.dev.write(endpoint, buffer, self.__claimed_interface, timeout)
  File "/home/tormod/pyusb/usb/core.py", line 630, in write
    _interop.as_array(data),
  File "/home/tormod/pyusb/usb/_interop.py", line 135, in as_array
    a.fromstring(data)
TypeError: must be string or read-only buffer, not list

I have git-bisected to confirm that it was commit e535b29e that broke this.

walac, 2016-06-18:
Hrm, thanks for reporting the regression. It might take some time for me to fix it, however, since my test hardware died.

tormodvolden, 2017-04-16:
@walac what kind of test hardware do you need? If you need a simple device that e.g. echoes bulk transfers I might be able to help out.

walac, 2017-04-17:
@tormodvolden I haven't had much time to look at PyUSB, but I am looking for a virtualized that can enable me run tests on travis.

jonasmalacofilho, 2020-01-05:
I think that supporting this (in 2020) would lead to a very ambiguous API, because of Python 3 vs. Python 2 differences.

First, now/in Python 3 this is a list of Unicode strings:

```py
buffer = ['\x07', '\xb8', '\x0b', '\x00', '\x00']
```

Presumably, the OP wanted to send the following bytes:

```py
>>> buffer = b'\x07\xb8\x0b\x00\x00'
>>> array.array('B', buffer).tobytes()
b'\x07\xb8\x0b\x00\x00'
```

But (if I'm not mistaken) the code before e535b29ef38246e7ceac2c97ac2078c2b430c89f would actually result in:

```py
>>> buffer = ['\x07', '\xb8', '\x0b', '\x00', '\x00']
>>> array.array('u', buffer).tobytes()
b'\x07\x00\x00\x00\xb8\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

The ambiguity comes from having `> 0x7f` escapes in strings: `\xb8` in a Python 3 string is interpreted as a Unicode character code 0xb8, instead of a raw 0xb8 byte.

_(and we should remember that as of January 2020 Python 2 has reached EOL)_

---

TL;DR.  I'm not sure what would be a good general behavior when dealing with lists of strings.  They could come from those porting old Python 2 code, or from real Unicode strings... and how should we handle non-ascii escapes in those strings?

I just don't thing this is worth pursuing, and would like to close this in a few days if there are no good counter-points (and a sane rule for the above ambiguity).

tormodvolden, 2020-01-07:
The calling code in question is 12 years old library code and I was happy about not looking into it as long as the legacy module was working (having reverted e535b29). I don't know why it was using lists of strings or "characters" in the first place - maybe it was made by looking at examples using strings or something else about pyusb 0.x at the time.

With the unavoidable migration to Python 3 coming up on the horizon, I dug into it last month and converted it to using lists of bytes instead, which makes fully sense anyway. So it now works fine with both 2 and 3 on stock pyusb. After further cleanup it became clear that porting it to pyusb 1.0 API was not difficult either, so now I hope we are set for another decade.

So I am OK with closing this issue, not having incentive to investigate any further on it, which I understand from your analysis (thanks!) would be difficult anyway.

jonasmalacofilho, 2020-01-09:
That seems fair.  Thanks again.

closed 2020-01-09 by jonasmalacofilho