pyusb/pyusb #151: Travis CI + pyusb

stephengroat, 2016-08-08:
I know that this isn't really an issue/bug, but I was wondering if I could get some help.

I'm looking to override pyusb for Travis CI/any other CI environment. When i call pyusb in travis there's (obviously) a problem:

````
Traceback (most recent call last):
  File "/home/travis/build/stephengroat/PyPowerUSB/test/test.py", line 7, in test_constructor
    handle = pwrusb.PyPwrUSB()
  File "pwrusb/pwrusb.py", line 20, in __init__
    busses = usb.busses()
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/usb/legacy.py", line 353, in busses
    sorted(core.find(find_all=True), key=lambda d: d.bus),
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/usb/core.py", line 1246, in device_iter
    for dev in backend.enumerate_devices():
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/usb/backend/libusb0.py", line 440, in enumerate_devices
    _check(_lib.usb_find_busses())
  File "/home/travis/virtualenv/python2.7.9/lib/python2.7/site-packages/usb/backend/libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
USBError: [Errno None] couldn't opendir(): No such file or directory
````

I was wondering if there's any single point that's easy to override for actions like these (to write action to a file for checking or comparing to known good values).

Thanks and sorry to bother (close if too off topic)

walac, 2016-08-10:
It feels like travis server/container doesn't enable usbfs drivers. You can write your own backend that would fake a real device.

stephengroat, 2016-08-10:
Are you talking about putting some in `/dev/usb` and having that device pipe something out to the file system?

walac, 2016-08-12:
I mean, implementing a mock [IBackend](https://github.com/walac/pyusb/blob/master/usb/backend/__init__.py) interface.

stephengroat, 2016-08-16:
perfect info. i'll try that this week

jonasmalacofilho, 2019-12-12:
Closing as there isn't much we can do on our side.

Users running into this problem – on systems without USB drivers or, more likely, in containers – should mock the necessary backends (see: ap--/python-seabreeze@f7e7d642f423).

And this is generally not a problem anymore: I routinely have tests run `find` in CI servers without issues (admittedly not on Travis, but that should work as well as long as containers are _not_ used).

closed 2019-12-12 by jonasmalacofilho