pyusb/pyusb #153: '_libusb_device_descriptor' object has no attribute 'port_numbers'

joashley, 2016-08-16:
Hello!

I'm new to python, so I apologize for any obvious errors. I'm trying to run a Formlabs Form 1 printer through their OpenFL python script (https://github.com/Formlabs/OpenFL). My code is as follows: 

```
import sys
import os
sys.path.append(os.path.abspath('../OpenFL'))

#from OpenFL import FLP
import FLP as FLP

import usb.backend.openusb as openusb
import usb.backend.libusb10 as libusb10
import usb.backend.libusb01 as libusb01
import OpenFL.Printer as P

p = P.Printer()
p.initialize()
p.start_printing(0)
```

I get an error saying that in core.usb, '_libusb_device_descriptor' object has no attribute 'port_numbers'. See attachment for full traceback. 

Note that if I do not manually import libusb10, I get a backend error saying that 'module' object has no attribute backend.

Any assistance would be greatly appreciated.

Ashley

[image]

walac, 2016-09-24:
Hi, what's the libusb version you have?

mcuee, 2018-09-19:
Closing old issues. Please reopen when new info provided.

closed 2018-09-19 by mcuee