pyusb/pyusb #159: Segmentation fault - double free or corruption (!prev)

WGH-, 2016-10-06:
So my test script simply does `ep.write(b'\x88' * 1)` inside a while True loop (iso transfer).

Crash usually occurs at exit, but the program may crash at its own will. Most of the time it's SIGSEGV or SIGBUS, and no stack trace is printed.

<pre>
Traceback (most recent call last):
  File "./debug_client.py", line 34, in <module>
    main()
  File "./debug_client.py", line 30, in main
    ep.write(b'\x88' * 1)
  File "/usr/lib64/python3.4/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/usr/lib64/python3.4/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/usr/lib64/python3.4/site-packages/usb/backend/libusb1.py", line 856, in iso_write
    return handler.submit(self.ctx)
  File "/usr/lib64/python3.4/site-packages/usb/backend/libusb1.py", line 674, in submit
    _check(_lib.libusb_handle_events(ctx))
KeyboardInterrupt
*** Error in `python3': double free or corruption (!prev): 0x0000000000ff8900 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x72443)[0x7f5e79243443]
/lib64/libc.so.6(+0x77d36)[0x7f5e79248d36]
/lib64/libc.so.6(+0x7853e)[0x7f5e7924953e]
/usr/lib64/libpython3.4.so.1.0(+0x99740)[0x7f5e79822740]
/usr/lib64/python3.4/lib-dynload/_ctypes.cpython-34.so(+0x7417)[0x7f5e77e42417]
/usr/lib64/python3.4/lib-dynload/_ctypes.cpython-34.so(+0x7429)[0x7f5e77e42429]
/usr/lib64/libpython3.4.so.1.0(+0xb20ff)[0x7f5e7983b0ff]
/usr/lib64/libpython3.4.so.1.0(+0x99717)[0x7f5e79822717]
/usr/lib64/libpython3.4.so.1.0(+0xb1b92)[0x7f5e7983ab92]
/usr/lib64/libpython3.4.so.1.0(+0x14a142)[0x7f5e798d3142]
/usr/lib64/libpython3.4.so.1.0(+0x14a89b)[0x7f5e798d389b]
/usr/lib64/libpython3.4.so.1.0(PyGC_Collect+0x28)[0x7f5e798d3d18]
/usr/lib64/libpython3.4.so.1.0(Py_Finalize+0x5f)[0x7f5e798bb1df]
/usr/lib64/libpython3.4.so.1.0(Py_Main+0x694)[0x7f5e798d1cf4]
python3(main+0x166)[0x400b06]
/lib64/libc.so.6(__libc_start_main+0xf0)[0x7f5e791f1630]
python3(_start+0x29)[0x400ba9]
======= Memory map: ========
00400000-00401000 r-xp 00000000 08:02 1998574                            /usr/bin/python3.4
00601000-00602000 r--p 00001000 08:02 1998574                            /usr/bin/python3.4
00602000-00603000 rw-p 00002000 08:02 1998574                            /usr/bin/python3.4
00d7d000-0101a000 rw-p 00000000 00:00 0                                  [heap]
7f5e70000000-7f5e70021000 rw-p 00000000 00:00 0 
7f5e70021000-7f5e74000000 ---p 00000000 00:00 0 
7f5e75854000-7f5e75869000 r-xp 00000000 08:02 13780942                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.3/libgcc_s.so.1
7f5e75869000-7f5e75a69000 ---p 00015000 08:02 13780942                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.3/libgcc_s.so.1
7f5e75a69000-7f5e75a6a000 r--p 00015000 08:02 13780942                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.3/libgcc_s.so.1
7f5e75a6a000-7f5e75a6b000 rw-p 00016000 08:02 13780942                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.3/libgcc_s.so.1
7f5e75a6b000-7f5e75a6c000 ---p 00000000 00:00 0 
7f5e75a6c000-7f5e7626c000 rw-p 00000000 00:00 0 
7f5e7626c000-7f5e76282000 r-xp 00000000 08:02 13279367                   /lib64/libusb-1.0.so.0.1.0
7f5e76282000-7f5e76482000 ---p 00016000 08:02 13279367                   /lib64/libusb-1.0.so.0.1.0
7f5e76482000-7f5e76483000 r--p 00016000 08:02 13279367                   /lib64/libusb-1.0.so.0.1.0
7f5e76483000-7f5e76484000 rw-p 00017000 08:02 13279367                   /lib64/libusb-1.0.so.0.1.0
7f5e764d1000-7f5e764d3000 r-xp 00000000 08:02 1992911                    /usr/lib64/python3.4/lib-dynload/_random.cpython-34.so
7f5e764d3000-7f5e766d3000 ---p 00002000 08:02 1992911                    /usr/lib64/python3.4/lib-dynload/_random.cpython-34.so
7f5e766d3000-7f5e766d4000 r--p 00002000 08:02 1992911                    /usr/lib64/python3.4/lib-dynload/_random.cpython-34.so
7f5e766d4000-7f5e766d5000 rw-p 00003000 08:02 1992911                    /usr/lib64/python3.4/lib-dynload/_random.cpython-34.so
7f5e766d5000-7f5e766ea000 r-xp 00000000 08:02 3146151                    /lib64/libz.so.1.2.8
7f5e766ea000-7f5e768e9000 ---p 00015000 08:02 3146151                    /lib64/libz.so.1.2.8
7f5e768e9000-7f5e768ea000 r--p 00014000 08:02 3146151                    /lib64/libz.so.1.2.8
7f5e768ea000-7f5e768eb000 rw-p 00015000 08:02 3146151                    /lib64/libz.so.1.2.8
7f5e768eb000-7f5e76afe000 r-xp 00000000 08:02 14177224                   /usr/lib64/libcrypto.so.1.0.0
7f5e76afe000-7f5e76cfd000 ---p 00213000 08:02 14177224                   /usr/lib64/libcrypto.so.1.0.0
7f5e76cfd000-7f5e76d1b000 r--p 00212000 08:02 14177224                   /usr/lib64/libcrypto.so.1.0.0
7f5e76d1b000-7f5e76d27000 rw-p 00230000 08:02 14177224                   /usr/lib64/libcrypto.so.1.0.0
7f5e76d27000-7f5e76d2b000 rw-p 00000000 00:00 0 
7f5e76d38000-7f5e76d78000 rw-p 00000000 00:00 0 
7f5e76d78000-7f5e76d7d000 r-xp 00000000 08:02 1992966                    /usr/lib64/python3.4/lib-dynload/_hashlib.cpython-34.so
[... 72 lines cut ...]
7f5e79783000-7f5e79784000 r--p 00016000 08:02 13332308                   /lib64/libpthread-2.22.so
7f5e79784000-7f5e79785000 rw-p 00017000 08:02 13332308                   /lib64/libpthread-2.22.so
7f5e79785000-7f5e79789000 rw-p 00000000 00:00 0 
7f5e79789000-7f5e7999a000 r-xp 00000000 08:02 1992809                    /usr/lib64/libpython3.4.so.1.0
7f5e7999a000-7f5e79b9a000 ---p 00211000 08:02 1992809                    /usr/lib64/libpython3.4.so.1.0
7f5e79b9a000-7f5e79b9c000 r--p 00211000 08:02 1992809                    /usr/lib64/libpython3.4.so.1.0
7f5e79b9c000-7f5e79bfe000 rw-p 00213000 08:02 1992809                    /usr/lib64/libpython3.4.so.1.0
7f5e79bfe000-7f5e79c19000 rw-p 00000000 00:00 0 
7f5e79c19000-7f5e79c3b000 r-xp 00000000 08:02 13334431                   /lib64/ld-2.22.so
7f5e79c76000-7f5e79cb6000 rw-p 00000000 00:00 0 
7f5e79cd6000-7f5e79ce5000 r-xp 00000000 08:02 13509247                   /lib64/libbz2.so.1.0.6
7f5e79ce5000-7f5e79ce7000 rw-p 0000e000 08:02 13509247                   /lib64/libbz2.so.1.0.6
7f5e79ce7000-7f5e79dec000 rw-p 00000000 00:00 0 
7f5e79def000-7f5e79df0000 rw-p 00000000 00:00 0 
7f5e79df0000-7f5e79df2000 rwxp 00000000 00:00 0 
7f5e79df2000-7f5e79e32000 rw-p 00000000 00:00 0 
7f5e79e32000-7f5e79e39000 r--s 00000000 08:02 12583044                   /usr/lib64/gconv/gconv-modules.cache
7f5e79e39000-7f5e79e3a000 rw-p 00000000 00:00 0 
7f5e79e3a000-7f5e79e3b000 r--p 00021000 08:02 13334431                   /lib64/ld-2.22.so
7f5e79e3b000-7f5e79e3c000 rw-p 00022000 08:02 13334431                   /lib64/ld-2.22.so
7f5e79e3c000-7f5e79e3d000 rw-p 00000000 00:00 0 
7ffdec467000-7ffdec488000 rw-p 00000000 00:00 0                          [stack]
7ffdec5b6000-7ffdec5b8000 r--p 00000000 00:00 0                          [vvar]
7ffdec5b8000-7ffdec5ba000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]</pre>

jonasmalacofilho, 2019-12-12:
Hi @WGH- 

It's been a long time... can you confirm that this still reproduces with PyUSB 1.0.2 (or git) and libusb 1.0.23?

mcuee, 2019-12-19:
I think we can close this in a week time if not hearing from the user.

WGH-, 2019-12-25:
Lol, this's is weird, I don't think I changed anything in either script or device firmware.  It does seem that the endpoint with zero `wMaxPacketSize` is selected.

```
Traceback (most recent call last):
  File "debug_client.py", line 34, in <module>
    main()
  File "debug_client.py", line 30, in main
    ep.write(b'\x88' * 1)
  File "/usr/lib64/python3.6/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/usr/lib64/python3.6/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/usr/lib64/python3.6/site-packages/usb/backend/libusb1.py", line 855, in iso_write
    handler = _IsoTransferHandler(dev_handle, ep, data, timeout)
  File "/usr/lib64/python3.6/site-packages/usb/backend/libusb1.py", line 650, in __init__
    packet_count = int(math.ceil(float(length) / packet_length))
ZeroDivisionError: float division by zero
Traceback (most recent call last):
  File "/usr/lib64/python3.6/weakref.py", line 624, in _exitfunc
    f()
  File "/usr/lib64/python3.6/weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "/usr/lib64/python3.6/site-packages/usb/_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "/usr/lib64/python3.6/site-packages/usb/_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "/usr/lib64/python3.6/site-packages/usb/backend/libusb1.py", line 667, in _finalize_object
    _lib.libusb_free_transfer(self.transfer)
AttributeError: '_IsoTransferHandler' object has no attribute 'transfer'
```

WGH-, 2019-12-25:
I'm not giving up, but please feel free to close the issue while I'm struggling with my old reproduction setup. I think I'll have time after New Year.

jonasmalacofilho, 2019-12-27:
I'll just leave it open for another 3 weeks.

There seems to be an issue here.  Maybe it's not on the PyUSB side, but still, I would like to know more about it.

WGH-, 2020-01-09:
Okay, apparently, I'm hitting the zero division because of https://github.com/libusb/libusb/issues/36. It's still a mystery why it hadn't surfaced before, though... 

Anyway, I tried both old PyUSB and libusb, and was unable to reproduce the bug.

I'd say let's close this, and if I ever stumble upon this bug again, I'll certainly bring better details and debugging on the table than the original post.

jonasmalacofilho, 2020-01-09:
Ok, that seems fair.  And thanks again for looking into this.

mcuee, 2021-07-30:
From https://github.com/libusb/libusb/issues/36#issuecomment-573674092
@WGH commented:
> @mcuee the pyusb issue is about completely different problem, though.
> 
> However, pyusb is indeed affected by this issue, as iso transfer codepath unconditionally calls `libusb_get_max_iso_packet_size` to calculate packet lengths, so if calculation is wrong, it breaks iso transfer. If the default AlternateSetting has zero `wMaxPacketSize`, it even leads to zero division.
> 
> https://github.com/pyusb/pyusb/blob/3bab67724ac4dc1d19f421abd302233314789b85/usb/backend/libusb1.py#L637-L656

mcuee, 2021-07-30:
Just come across this comment when I was reviewing the libusb issues. Maybe we need to reopen this.

jonasmalacofilho, 2021-07-30:
Unless I'm missing something, I agree that

> the pyusb issue is about completely different problem, though.

By this I mean: it's not the segfault this issue was originally about.

So a new issue would be more appropriate... or maybe don't need to track it at all?

In general, every upstream issue has a high probably of affecting us anyway.  And unless the problem specifically affects PyUSB (e.g. requires a fix or workaround in PyUSB) or PyUSB users (e.g. not generally hit by other users), I don't think we need to track it.

mcuee, 2021-07-30:
Good point. A new issue is more appropriate. 

As far as I see, I have not seen much reports of the isoc transfer with pyusb, Normally isoc needs to work with async model which is not implemented in pyusb yet. I am actually not sure how well pyusb works with isoc transfer. It did not work for me based on my testing.

closed 2021-07-30 by mcuee

mcuee, 2021-07-30:
Just for reference: isochronous transfer APIs under libusb are grouped together with the other async I/Os.
https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html

mcuee, 2021-07-30:
On the other hand, isoc is not so easy to get working properly. And certainly libusb Windows has quite some issues as well. So yes we should close this issue. I will create a new issue with isoc transfer if pyusb does not work yet libusb C application works with a known good FW.

mcuee, 2021-08-03:
My own testing results in #159 actually show that pyusb isoc codes seem to work well, provides that the FW is working, and provided it is not under Windows since libusb-1.0 Windows does not seem to work well with isoc transfer.