pyusb/pyusb #161: GPG signatures for source validation

NicoHood, 2016-11-22:
As we all know, today more than ever before, it is crucial to be able to trust our computing environments. One of the main difficulties that package maintainers of Linux distributions face, is the difficulty to verify the authenticity and the integrity of the source code.

The Arch Linux team would appreciate it if you would provide us GPG signatures in order to verify easily and quickly of your source code releases.

**Overview of the required tasks:**
* Please [create and/or use a 4096-bit RSA keypair](https://wiki.archlinux.org/index.php/GnuPG#Create_key_pair) for the file signing
* The key may be used for [email encryption](https://www.enigmail.net/index.php/en/) too
* Keep your key secret, use a strong unique passphrase for the key
* [Sign every new commit by default](https://help.github.com/articles/signing-commits-using-gpg/)
* [Sign new tags](https://help.github.com/articles/signing-tags-using-gpg/)
* Verify the Github Release tarball download against your local source
* [Sign Github Release tarballs](https://wiki.debian.org/Creating%20signed%20GitHub%20releases)

**Additional Information: **
* https://help.github.com/categories/gpg/
* https://git-scm.com/book/en/v2/Git-Tools-Signing-Your-Work https://www.qubes-os.org/doc/verifying-signatures/
* https://developers.google.com/web/fundamentals/security/encrypt-in-transit/why-https

Thanks.

walac, 2016-12-14:
Sorry for the delay, I was in vacation. Will work on this.

NicoHood, 2017-06-15:
@walac any progress on that? `twine` creates beautiful signed releases on pypi (-i switch) and `gpgit` does the same for Github releases along with a general GPG guide. It would be nice if you can also sign this python module.

Cheers
Nico

walac, 2017-06-18:
Hi, I configured signed commits some time ago, but didn't make any since then. Quite busy semester, left me a big backlog to handle, hopefully starting in July when things settle a bit in work.

NicoHood, 2017-06-18:
I understand. You can sign an older release also with GPGit. Just run it with the same version tag and it will resign the specified tag :) Do it when you got time, but especially for the next release it is of high importance. Thanks

NicoHood, 2017-10-10:
@walac The latest release 1.0.1 is not tagged on Github, nor GPG signed on pypi. Can you please tag it along with GPG signatures? Thanks.

walac, 2017-10-11:
Sorry, it was pushed to by someone else to fix #180

walac, 2017-10-11:
Notice that signing every commit is unfeasible because of code comming from contributors and merge commits.

rnixx, 2017-10-11:
@walac @NicoHood I did no git tagging because of missing write access. @walac may you be so kind and add a tag?

rnixx, 2017-10-11:
@walac oh, just seen you already tagged. sorry for the noise

NicoHood, 2017-10-11:
@walac it will be sufficient if you provide us with a gpg signed tarball. This can be done with twine using the -s option if i remember correct. You need to make sure that the source of the tarball is whats from the git of course. You can also use my tool gpgit or do the signing manual. Its simple:
https://github.com/NicoHood/gpgit

walac, 2017-10-11:
@NicoHood I wrote some code yesterday night to automate this, will push a 1.0.2 version with signed tarball soon (hopefully until this weekend).

I need to look at gpgit, it might not only be useful for PyUSB, for for things on my job too :)

NicoHood, 2017-11-25:
@walac Thanks! I have updated the package with the signature of the fingerprint `B04841AE800C1BF01FE1BC3D084C5584542E1574`.

closed 2017-11-25 by NicoHood