pyusb/pyusb #162: Class control transfers not working as expected with WinUSB + libusb1

c1728p9, 2016-11-26:
When using WinUSB and libusb1 on Windows class requests sent for interfaces other than 0 don't work correctly. In the case below, a class out request intended for interface 3 goes to interface 0 when using WinUSB + libusb1. When using Ubuntu 16.04 running in a VM on the same machine the request goes to interface 3 as expected.

Configuration information:
Windows 7 x64
libusb-1.0.20
zadig_2.2 used to install WinUSB (v6.1.7600.16385)
USB device used was a [Switch-Science-mbed-LPC824](https://developer.mbed.org/platforms/Switch-Science-mbed-LPC824/) running [DAPLink](https://github.com/mbedmicro/daplink)
Python script the test was run with: [script.zip](https://github.com/walac/pyusb/files/614042/script.zip)

USB capture on Windows (bad):
[image]

USB capture on Ubuntu (good):
[image]

mcuee, 2016-12-14:
Not so sure if this has something to do with WinUSB limitation. You can want to switch to libusbk driver to see if that helps.

https://github.com/libusb/libusb/wiki/Windows#known-restrictions

"With WinUSB, when LIBUSB_RECIPIENT_INTERFACE is used for the transfer, the WinUSB DLL forces the low byte of wIndex to the interface number, regardless of what you set it to. "

c1728p9, 2016-12-14:
Thanks for the response @mcuee. This looks like the issue I'm having. The link with further details about WinUSB - "http://www.lvr.com/forum/index.php?topic=331" doesn't work for me though. I'll try libusbk as an alternative.

closed 2016-12-14 by walac

mcuee, 2016-12-30:
BTW, I have updated the Wiki to reflect the fact that this is not a real limitation. I have also provided the link for the explanation. http://www.osronline.com/showthread.cfm?link=223812