pyusb/pyusb #164: "IndexError: array index out of range" when reading product property

c1728p9, 2016-12-06:
When reading the product property of a device I occasionally get an array index out or range error. This is on Ubuntu 16.04 when using libusb1. More specifically, I'm seeing this when running usb.core.find using a custom_match which reads the product property.

Exception:
```
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 841, in product
    self._product = util.get_string(self, self.iProduct)
  File "/usr/local/lib/python2.7/dist-packages/usb/util.py", line 330, in get_string
    return buf[2:buf[0]].tostring().decode('utf-16-le')
IndexError: array index out of range
```

c1728p9, 2016-12-06:
I took a look at the USB traffic when this occurs. A zero length IN transaction is sent in response to the GetDescriptor request during the data stage. If there is a check that "len(buf) > 0" then the IndexError could be prevented.

walac, 2016-12-14:
Hrm, a more elegant message error is indeed desirable, thanks for the report.

commit 3bab6772 (2020-01-12): Raise a USBError for obviously invalid descriptors

In certain scenarios some devices may return zero-length descriptors.
However, all descriptors must have at least two bytes (the descriptor
length in bytes and the descriptor type), and the USB 3.2 specification
states that:

> If a descriptor returns with a value in its length field that is less
> than defined by this specification, the descriptor is invalid and should
> be rejected by the host.

Instead of returning invalid descriptors, modify get_descriptor so it
raises a proper USBError as soon as/if the descriptor is found to be
invalid. [... cut]
closed 2020-01-12 by jonasmalacofilho