pyusb/pyusb #165: filename not work  and not list all device attach with usb

piyushgohel, 2016-12-21:
Hello,

I am working with pyusb but it will not return device name connected  with usb. It also not return all device list.Is there any event in pyusb which identify plug and unplug event.

import usb
		busses = usb.busses()
		for bus in busses:
		    devices = bus.devices
		    for dev in devices:
		    	print dev.filename

		        print "Device:", dev.filename
		        print "  idVendor: %d (0x%04x)" % (dev.idVendor, dev.idVendor)
		        print "  idProduct: %d (0x%04x)" % (dev.idProduct, dev.idProduct)

walac, 2016-12-21:
No, hotplug events are on going. You may want to switch to version 1.0 api.

piyushgohel, 2016-12-22:
I am using pyusb  v1.0.0

jonasmalacofilho, 2019-12-12:
Filenames are no longer supported: the legacy API returns empty strings (and libusb-compat fakes them with the address number).  For hotplug events, see #64 ("Implement hotplug events").

The last remaining problem – "It also not return all device list" – would require some more information to be confirmed, and could be explained by the use of legacy APIs and/or backends.

If this problem still reproduces with PyUSB git (or at least 1.0.2) and libusb 1.0.x, please open a new issue including the output after setting the `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` environment variables.

closed 2019-12-12 by jonasmalacofilho