pyusb/pyusb #166: pyusb script not work on MAC

piyushgohel, 2016-12-21:
Hello,

I have written pyusb script. But it will not work on MAC. It will give following error.

 busses = usb.busses()
  File "/Library/Python/2.7/site-packages/usb/legacy.py", line 353, in busses
    sorted(core.find(find_all=True), key=lambda d: d.bus),
  File "/Library/Python/2.7/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

==========================================================================
My code is as follow:

    import usb
    busses = usb.busses()
    for bus in busses:
    	devices = bus.devices
    	for dev in devices:
    		print dev.dev
    		#print usb.util.get_string(dev, 256, dev.iManufacturer)
    		#print usb.util.get_langids(dev)
        	print "Device:", dev.filename
        	print "  idVendor: %d (0x%04x)" % (dev.idVendor, dev.idVendor)
        	print "  idProduct: %d (0x%04x)" % (dev.idProduct, dev.idProduct)

walac, 2016-12-21:
Do you have a version of libusb up and running?

mcuee, 2016-12-30:
FYI, pyusb works fine under Mac OS X for me (macOS Sierra 10.12.2 now).

SyntaxChris, 2017-01-26:
@piyushgohel if you have home brew installed run `brew install libusb`.  That should do it.

John-K, 2017-08-07:
You likely don't have permission. Running your script with sudo may fix the issue