pyusb/pyusb #173: prevent SET_CONFIGURATION from being sent when using set_configuration with libusb0

ReallyCoolName, 2017-03-23:
It's a small use case, but at the very least maybe one day I will spare someone the hours of work and googling I've spent on this, even if this isn't resolved (workaround at the end):

The problem, I suspect, is [mentioned here](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga186593ecae576dad6cd9679f45a2aa43) - if the device is already configured with the selected configuration, it might reset, and in the case of my android smartphone (ever since the upgrade to Marshmallow - before that I had no problems), it does.

[If you're actually interested in why this is a problem, continue at your own peril. If not, I just want to thank @walac  for PyUSB. It spares me a **_lot_** of headache!]

```
import usb
from usb.backend import libusb1
from usb.backend import libusb0
be = libusb1.get_backend()
from time import sleep

dev = list(usb.core.find(find_all=True,backend=libusb0.get_backend()))[0]
if not (dev.idProduct==0x2d00):
    dev.ctrl_transfer(0xc0,51,data_or_wLength=2)
    dev.ctrl_transfer(0x40,52,wIndex=0,data_or_wLength='SAMSUNG')
    dev.ctrl_transfer(0x40,52,wIndex=1,data_or_wLength='SAMSUNG_Android')
    dev.ctrl_transfer(0x40,52,wIndex=2,data_or_wLength='16DIGITSERIALNUM')
    dev.ctrl_transfer(0x40,52,wIndex=3,data_or_wLength='AOA')
    dev.ctrl_transfer(0x40,52,wIndex=4,data_or_wLength='Whatever')
    dev.ctrl_transfer(0x40,52,wIndex=5,data_or_wLength='4')
    dev.ctrl_transfer(0x40,53)
    sleep(2)
    dev = list(usb.core.find(find_all=True,backend=be))[0]
    dev.set_configuration()
```

the control transfers are part of the protocol to bring the android device into accessory mode, which allows the phone to communicate with (in this case) the computer (this provides some extra functionality which isn't possible when the device is using the other protocols). Problem is that when the device is disconnected (even programatically), it leaves the open accessory mode.

anyway, at the moment I'm switching to `libusb1` after the device has entered accessory mode (it has some problems of it's own - at mentioned in a previous issue, it doesn't handle root hubs very well, which is why I can't use it for the control transfers). Haven't fully adapted this solution yet, but the `read()` works, so

mcuee, 2018-04-11:
It seems to me right now dev.set_configuration() is compulsory under pyusb. This should be made optional for use cases like this.

jonasmalacofilho, 2019-12-12:
I think the [workaround](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration) mentioned in FAQ applies here.

> One solution to this behaviour is to consider the currently active configuration, as described in the [configuration selection and handling](http://libusb.sourceforge.net/api-1.0/libusb_caveats.html). "If the configuration we want is already active, then we don't have to select any configuration":
>
>```python
>try:
>    cfg = dev.get_active_configuration()
>except usb.core.USBError:
>    cfg = None
>if cfg is None or cfg.bConfigurationValue != cfg_desired:
>    dev.set_configuration(cfg_desired)
>```

Additionally, the soft reset behavior is part of the USB spec.

closed 2019-12-12 by jonasmalacofilho