pyusb/pyusb #174: About the use of pyusb in the multiprocess

wbb12, 2017-03-26:
I have to communicate with several identical devices by usb. I have communicated with one single device by pyusb with python3.5 successsfully.  however, when it comes to several devices, I  prepared to use multiprocesses to communicate with each devices respectively.  It seems the process will  block the others, and the data received from the device go wrong sometimes. 
is that a fact that the python programes with pyusb can only communicate with on single device at a particular time?

rec, 2017-03-30:
While I am not the writer of this program, I don't see how you expect this to work. 

There's only one wire.  How do you expect to send more than one message in parallel?

walac, 2017-04-02:
If they are identical, how do you distinguish them?

hardkrash, 2017-04-28:
Not that I know how wbb12 does this but I use bus and port numbers.

ldnelso2, 2019-01-02:
@hardkrash how do you use the bus and port numbers?  This is mentioned in the tutorial, but I can't figure out how to address them to write serial commands to multiple identical devices.

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee