pyusb/pyusb #176: pyusb does not install its dependencies with setup.py.

rec, 2017-03-30:
Hello!  Thanks for all your work on this useful library.

We'd like to be using it for our LED lighting system, [BiblioPixel](https://github.com/ManiacalLabs/BiblioPixel), but there's an issue that makes this impossible...

pyusb cannot be installed simply with setup.py.  There is an additional manual step that must be performed to install an appropriate binary library.

One of our key features is that our program can be installed only with setup.py.  This isn't just "nice to have" - it's essential because there are other packages that depend on ours, also using setup.py, and for all I know there are other packages that depend on those packages.

More, we have continuous integration (CI) testing, and that also requires that all packages that we depend on be automatically installed.

CI isn't just a fad - pretty well all production programs have it these days, and it repeatedly saves us from all sorts of stupid mistakes, mostly our own.  :-)

So having a completely automatic installation is a key feature for us.  

Please let us know!  We might be able to give you some help with accomplishing this, as we have some experience packaging Python libraries for distribution.

Thanks for your attention!

walac, 2017-04-02:
Installing `libusb` is definitely out of scope of PyUSB. If you do need to install setup.py, maybe you can provide a package that install libusb and marks pyusb as a dependency, install `pip`.

rec, 2017-04-02:
A package that doesn't install its dependencies is suboptimal, but we only have limited time in the day to deal with chores! 

Thanks, I'm closing this.

closed 2017-04-02 by rec