pyusb/pyusb #178: ValueError: NULL pointer access

erezposner, 2017-04-01:
I have two barcode readers.  I get a null pointer access when trying to init only one of them (i.e one works , the  seconds does not)
under windows i used Zadig to modify the driver to libusb-win32.
and uses this code below.
again, only one of the barcodes work. i saw that the one that is working only has one interface while the one that is not working has 2. i know my error description is a bit cryptic, but if you can direct me to any missing information i will send it asap.

    def InitUSB(self):
        # bc = USB_Device()
        HID_Interface_Class=3
        # Find DevicesList Filtered by idVendor and HIDInterfaceClass
        # devices = usb.core.find(find_all=True)
        devices = usb.core.find(find_all=True, custom_match=find_Device_InterfaceClass(HID_Interface_Class))

        # devices = usb.core.find(find_all=True, idVendor=0x13ba, custom_match=find_Device_InterfaceClass(HID_Interface_Class))#TODO Uncomment if having trouble

        if devices is None:
            raise ValueError('Our device is not connected')
        NumberOfDevicesFound = len(devices.gi_code.co_consts) - 1
        if NumberOfDevicesFound > 1:
            raise ValueError('More Than One Device is connected')

        dev = devices.next()

walac, 2017-04-02:
`find` function will call `libusb` initialization  functions, which probably is messing things up.

erezposner, 2017-04-02:
so what are suggesting i should do?

walac, 2017-04-02:
Do only one call `find` function to return both devices references.

erezposner, 2017-04-02:
I am only doing one find call. see other rows are commented.

walac, 2017-04-02:
So, where are you actually caling the second device?

erezposner, 2017-04-02:
Ok , i guess i wasnt clear. 

i only want to use the second barcode reader (libusbK). during development i used the first one. then i replaced the barcode reader due to its physical dimensions. I was under the assumption  that there won't be any difference between the two.

so what i am saying that i want the code that works with the first one to work with the second. but until now without luck

walac, 2017-04-03:
Hrm, you can use `PYUSB_DEBUG` and `LIBUSB_DEBUG` environment variables to spot exactly what's going on. It feels like a libusb issue.

erezposner, 2017-04-04:
I've been playing with it and now i can successfully see the connect to the second barcode.
Finally i understand the difference in behavior between the two barcode readers.

now i have a new problem, the second barcode is seem to operate both as an HID device and also handled by the pyUSB. doing a little digging i saw that the second one has two interface while the first has only one.
with the current code, as i scan a barcode with the second reader the first interface is being listened by the pyusb and the other interface is being handled by the kernel as an hid device.
(while using the first barcode reader the one interface is being listened by the pyusb and everything is working as expected)

so my question is, how can i disable the second interface? is there a way to listen to both interfaces? 
thank you

walac, 2017-04-04:
Can't you just ignore the hid interface?

erezposner, 2017-04-04:
i am not what you mean by ignoring. i have no control over it.
Basically, my app displays the user an HTML on chromium (I use Flask && chromium) , and in the background i have a thread that listens to the input from the barcode reader.
now the problem is that every time i scan a barcode it is as if i opened another tab in the chromium.
this is due to the second interface defined as HID.

i also get the value using the first interface && pyUSB.

walac, 2017-04-04:
by "ignore it" I mean just don't send any commands to it. I am afraid I didn't actually understand your issue.

erezposner, 2017-04-04:
Ok, i will explain.
i have barcode reader that has two interfaces:
[image]

Now, as far as the actual program it is as follows: the user sees on chrom an HTML like this:
[image]
what i want to do is to listen to any scan and handle it.

When i use barcode reader with only one interface everything works fine (using the code above) , i get the barcode data.

If i use barcode reader with two interfaces what happens is that i get the data from interface 0, and it seems that interface 1 is still bind to the kernel and acts as HID device resulting with this:
[image]

in the image you can see that the previous screen is gone because it as if i opened another tab and put the scanned number in the address bar.

i cannot control, or do not know how to control the actions performed from interface number 1.
how can i disable it?

mcuee, 2019-11-29:
Please use pyusb mailing list for technical support.

closed 2019-11-29 by mcuee