pyusb/pyusb #179: core.find() string filters requires write access to all connected devices

tormodvolden, 2017-04-16:
The currently resulting "langid" error message has been mentioned in #139 and issue #145 but the OPs there might have other original issues. However the description below should be clarifying for the other discussions as well.

The "langid" error message is misleading, however the underlying issue also causes core.find() to be less usable.

Background (facts):

1. Generally, to retrieve product, manufacturer and serial number /strings/ using USB requests you'll need appropriate permissions to the USB device (write permissions on USB device node on Linux).

You can see this with e.g. lsusb, with our without superuser rights:
lsusb -v | grep 'i[MPS]'
sudo lsusb -v | grep 'i[MPS]'

2. Before retrieving any "normal" string descriptors from a device, the code should request the "magic" langid string descriptor which contains values needed for correctly retrieving the "normal" ones. This is why, on failing permissions, the current error message mentions "langid", which can be confusing.

3. If you search for a device by filtering on one of these strings, e.g.
dev = usb.core.find(product='X') the code will want to see the product string of all connected USB devices until it finds a match. If this search passes through a device for which you have no permissions, retrieving the string will fail, and currently break the search. If you're lucky, you'll find your device before you reach a non-readable device. The order of devices depends on the OS, physical connection, or possibly other factors out of pyusb's control.

If you filter on idVendor and/or idProduct values there is no issue because in this case communication with the devices is not needed - these values are cached by the OS and are readable by anyone.

Solutions (IMO):

4. I think core.find() should ignore devices if it cannot read the appropriate value (i.e. device strings) from it. Currently it uses util.get_string() which will raise an exception if user permissions are not sufficient. 

A quick fix is for util.py:get_string() (at line 314) to return None or an empty string instead of raising the "langid" exception.

Definitely we would like people to be able to find their matching devices without having full write permission to all connected devices (raw access to USB drives etc).

5. On the other hand, if the user explicitly queries a device for a string without having proper permissions, get_string() should fail (or warn) and not just return None or an empty string.

In the example of lsusb, it prints warning messages if some devices cannot be opened.

Maybe warnings would be enough in both the core.find() and the explicit get_string() cases.

6. Otherwise we could handle the core.find() case differently, using a separate get_string() like function or have get_string() take a "None-on-error" option.

Note that core.py already has a _try_get_string() function which can be set to return None or an empty string on failing permissions.

Severity:

7. The misleading error message "langid" instead of "Access denied" is a relative recent regression, probably since commit 525d3b6be22b3ce2595c0f20a20ce7d4883176cb

On the other hand, the failing core.find() on missing permissions if filtering on strings I believe has been like this for a long time / ever.

mcuee, 2018-04-11:
This is interesting and may be related to other issues raised.

miphip, 2019-10-15:
This may also occur when a device has been opened/claimed by another process. At least I've seen this on Windows 10 with libusb-1.0 and a device using the winusb.sys driver. In this case libusb returns -3 (permission error).

jonasmalacofilho, 2020-01-05:
Extending on the possible errors: depending on the type of device and already bound driver, `libusb_open` might also return `LIBUSB_ERROR_NOT_SUPPORTED` (Operation not supported or unimplemented on this platform).

I have an initial patch that seems works, I'll push it once I'm able to test it a bit more.

---

_(For my personal notes: related to my `04d9:0a0e`, `04d9:0169` and `05e3:0607` composite devices -- Logitech sound bar and mouse)._

commit 0d53c415 (2020-01-07): Make find() robust against inaccessible attributes

Device attributes that require reading from the device, such as string
descriptors, can be inaccessible for several reasons (e.g. insufficient
permissions).

This is particularly relevant on Windows, where even with adequate
permissions it might not be possible to read from all devices, usually
due to driver limitations.

This patch makes find() tolerant to such issues, treating them as failed
matches instead of aborting with exceptions.  Internally, an `object()`
sentinel is used in place of the expected value, to prevent accidental
matches. [... cut]
closed 2020-01-07 by jonasmalacofilho