pyusb/pyusb #180: Building with recent buildout always breaks the first time because of capitals in project name

StefanRijnhart, 2017-05-04:
Building PyUSB from buildout 2.9.0 or higher breaks the first time. After that, buildout finishes completely. The problem seems to be a confusion between __pyusb__ and __PyUSB__ as project names.

```
Getting distribution for 'pyusb>=1.0.0b1'.
zip_safe flag not set; analyzing archive contents...
While:
  Installing odoo.
  Getting distribution for 'pyusb>=1.0.0b1'.

An internal error occurred due to a bug in either zc.buildout or in a
recipe being used:
Traceback (most recent call last):
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/buildout.py", line 1982, in main
    getattr(buildout, command)(args)
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/buildout.py", line 668, in install
    installed_files = self[part]._call(recipe.install)
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/buildout.py", line 1419, in _call
    return f()
  File "/home/buildout/anybox.recipe.odoo/anybox/recipe/odoo/base.py", line 1114, in install
    self.install_requirements()
  File "/home/buildout/anybox.recipe.odoo/anybox/recipe/odoo/base.py", line 557, in install_requirements
    eggs_recipe.install()
  File "/home/buildout/eggs/zc.recipe.egg-2.0.3-py2.7.egg/zc/recipe/egg/egg.py", line 126, in install
    reqs, ws = self.working_set()
  File "/home/buildout/eggs/zc.recipe.egg-2.0.3-py2.7.egg/zc/recipe/egg/egg.py", line 84, in working_set
    allow_hosts=self.allow_hosts)
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/easy_install.py", line 914, in install
    return installer.install(specs, working_set)
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/easy_install.py", line 666, in install
    for dist in self._get_dist(requirement, ws):
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/easy_install.py", line 564, in _get_dist
    dists = [_move_to_eggs_dir_and_compile(dist, self._dest)]
  File "/home/buildout/eggs/zc.buildout-2.9.0-py2.7.egg/zc/buildout/easy_install.py", line 1731, in _move_to_eggs_dir_and_compile
    assert newdist is not None  # newloc above is missing our dist?!
```

Earlier versions of buildout already blurred out this warning:
```
Installing PyUSB 1.0.0
Caused installation of a distribution:
pyusb 1.0.0
with a different project name.
Got pyusb 1.0.0.
```

The relevant change in buildout is https://github.com/buildout/buildout/commit/fbb8c37b9cd2e303c37528b4f3e488bceff47f1b#diff-784f9bb654bbcee96dcbc547bbddd493R1708

I honestly don't know enough about Python packaging to tell where this can be fixed best. Does it mean that the PyUSB distribution uses capitals in an inconsistent or non standard way? Or is the casing arbitrary and should zc.buildout be resilient to these changes?

sebastienbeau, 2017-05-27:
I am impacted too

ventilooo, 2017-06-26:
So am I :+1:

ventilooo, 2017-07-17:
@walac ping ?

walac, 2017-07-17:
Hi, very very sorry, I am catching up on the project soon. Please read details [here](https://sourceforge.net/p/pyusb/mailman/message/35900035/)

StefanRijnhart, 2017-07-17:
@walac take care. Your health is much more important.

walac, 2017-07-17:
@StefanRijnhart thank you for understanding

hbrunn, 2017-08-18:
I've just proposed #189 which seems to fix the issue for me. In the long run, I think either the package or the module should be renamed.

For now, you can say:

```
[buildout]
find-links =
    https://github.com/Therp/pyusb/tarball/master-rename-pyusb/pyusb-1.0.1.tar.gz
...
[versions]
pyusb = 1.0.1
```

in your buildouts to fix this.

StefanRijnhart, 2017-08-18:
Thanks @hbrunn! Unrelated to this issue, but for the Odoo people here: if you actually want to use USB communication from Odoo POS, you need do need 1.0.0b1 because of API changes in this package afterwards.

StefanRijnhart, 2017-08-18:
Let me post my own workaround:

```
[buildout]
extensions = gp.vcsdevelop
vcs-extend-develop =
    git+https://github.com/walac/pyusb@1.0.0b1#egg=pyusb
```

rnixx, 2017-10-09:
The problem is that some change in buildout or setuptools requires tarballs uploaded to pypi simple index needs to case sensitive match package name in setup.py. Tarball on pypi is named PyUSB..., but package name in setup is pyusb, so this fails.

My workaround right now is a local release of the original package. I hope all the inconsistencies from the past are gone once all the stuff uses new pypi.org, but progress seems slow.

@walac Wish you all the best and hope you get better soon. If i can have persissions on pypi i'll drop a new tarball.

rnixx, 2017-10-09:
@walac Ah, my user on pypi is rnix

walac, 2017-10-09:
@rnixx I added you as a maintainer, let me know if you need something else

rnixx, 2017-10-09:
@walac Thank you! I needed to bump the version in order to upload a new tarball. pypi changed the policy that it's no longer possible to simply replace the tarball for an existing version.

The related pull request is here https://github.com/walac/pyusb/pull/192

I can merge on my own if i get write access here too

Thanks for dropping in

jonasmalacofilho, 2019-12-11:
@StefanRijnhart 

> Unrelated to this issue, but for the Odoo people here: if you actually want to use USB communication from Odoo POS, you need do need 1.0.0b1 because of API changes in this package afterwards.

Please give more details about this issue, it seems like it could be relevant.

---

On the original issue, providing tarballs that match the case of the project name should have solved this issue (done since 1.0.1).

closed 2019-12-11 by jonasmalacofilho

StefanRijnhart, 2019-12-11:
That would be the dreaded "Device has no langid" error if I remember correctly. Is what Odoo mention themselves: https://github.com/odoo/odoo/blob/12.0/addons/point_of_sale/tools/posbox/overwrite_before_init/etc/init_posbox_image.sh#L111

jonasmalacofilho, 2019-12-11:
Unfortunately there are many possible causes for that.  Well, we'll tackle some of them in #139 and #179, and hopefully they'll also cover the problems in odoo.

StefanRijnhart, 2019-12-11:
That's great! Thanks for your help.