pyusb/pyusb #181: Please generate HTML docs!

fake-name, 2017-05-09:
Please, please, PLEASE generate HTML docs for PyUSB.

The useability of pydoc is complete crap compared to a web page - With a web page, I can easily have multiple tabs open to different sections of the documentation, there is an actual navigation interface, etc... While PyDoc works, it's minimal at best, and straight out of the 80's at worst.

PyDoc can [natively generate somewhat ugly, but quite functional documentation](https://github.com/fake-name/ChromeController/blob/master/gen_doc.sh#L26-L33), and github can do the hosting.

mcuee, 2021-08-24:
This seems to work fine.

```
(py39venv) C:\work\libusb\pyusb [master ≡ +2 ~4 -0 !]> .\pydoc.bat

C:\work\libusb\pyusb>python -m pydoc -w usb
wrote usb.html

C:\work\libusb\pyusb>python -m pydoc -w usb.backend
wrote usb.backend.html

C:\work\libusb\pyusb>python -m pydoc -w usb.core
wrote usb.core.html

C:\work\libusb\pyusb>python -m pydoc -w usb.control
wrote usb.control.html

C:\work\libusb\pyusb>python -m pydoc -w usb.util
wrote usb.util.html

C:\work\libusb\pyusb>python -m pydoc -w usb.legacy
wrote usb.legacy.html

C:\work\libusb\pyusb>python -m pydoc -w usb._debug
wrote usb._debug.html

C:\work\libusb\pyusb>python -m pydoc -w usb._interop
wrote usb._interop.html

C:\work\libusb\pyusb>python -m pydoc -w usb._lookup
wrote usb._lookup.html

C:\work\libusb\pyusb>python -m pydoc -w usb._objfinalizer
wrote usb._objfinalizer.html
```

fake-name, 2021-08-24:
I mean, yeah, but my point was stick that on github pages or something.

mcuee, 2021-08-25:
> I mean, yeah, but my point was stick that on github pages or something.

But there are many broken links in the generated HTML files. So it does not seem to be a simple task.

arist0v, 2023-03-25:
Sphinx could probably parse it and get better (looking and producing) documentation , you should take a look(still can be hosted on github pages)