pyusb/pyusb #182: Only a single error class for different error categories

fake-name, 2017-05-09:
Right now, PyUSB seems to have a single error class for basically all errors that can possibly be thrown.

This makes error handling quite irritating. For example, in my current case, I'm basically polling a device for new updates. As such, I normally call `read()` with a very small timeout, until I see a timeout. However, the timeout raises the exact same error as a hardware issue, like the hardware being disconnected.

It would make much more sense logically to have a more structured set of errors. Everything could inherit from USBError, so it wouldn't break the existing code.

Something like 

```
USBError()
     | ---- USBTimeoutError()
     | ---- USBPermissionError()
     \ ---- USBHardwareError() 
```

With that structure, I can just catch `USBTimeoutError()`. Right now, I have to catch every incidence of `USBError()`, and conditionally re-raise the exception based on the error number.

Querela, 2019-12-29:
Would all this be a `USBHardwareError`? Would we need an hardware error or does the generic error suffice?
```python
LIBUSB_ERROR_IO
LIBUSB_ERROR_NO_DEVICE
LIBUSB_ERROR_NOT_FOUND
LIBUSB_ERROR_BUSY
LIBUSB_ERROR_OVERFLOW
LIBUSB_ERROR_PIPE
LIBUSB_ERROR_INTERRUPTED
LIBUSB_ERROR_NO_MEM
```
http://libusb.sourceforge.net/api-1.0/group__libusb__misc.html#gab2323aa0f04bc22038e7e1740b2f29ef

But a timeout error would be great.

jonasmalacofilho, 2020-01-05:
For what is worth, I think/agree that we should only have a handful of specialized exceptions, for the more common (and well defined) failure cases; timeouts are a great example of this.

Querela, 2020-01-05:
I started with it:
  https://github.com/pyusb/pyusb/compare/master...Querela:issue-182

but I found that the Backend libusb0 does not have any error number constants defined ...
For openusb, it would be trivial to mirror it like for libusb1.

I guess an Access and Timeout error would be good. The other types were harder to place ...

jonasmalacofilho, 2020-01-07:
I think that errors that we don't (yet) handle in a special way should remain as `USBError`; I see no point in a generic `USBHardwareError` subclass.

Besides that, any specialized errors that we introduce should be comprehensive (by this I mean: `except USBTimeoutError:` should catch all timeout-type errors).

This should be easy for timeouts, but correctly tagging all access/permission issues might be significantly harder.  At least we need to fix things like differentiating between the many possible causes for missing langids when trying to get string descriptors.  And we also need to check if libusb can reliably tell permission errors from other issues (which is generally a hard problem).

Querela, 2020-01-07:
Well, then at first only `USBTimeoutError` because that can see some real use and is fairly distinct and easy to implement.
As far as I saw in the source code, timeouts can only happen in the native backend libraries and are passed through `_check(ret)`. In `libusb0` there are no error codes defined, so I would leave it out.

If other errors types become more defined, we can do them in another Pull request.
But as I'm not really sure what belongs together I can't really implement anything.
Access errors seemed fairly distinct with its error code definition but I'm no expert ...

Querela, 2020-01-07:
libusb0 uses native IO error codes?
- https://docs.python.org/3/library/errno.html

Timeout:
- `errno.ETIME` (?)
- `errno.ETIMEDOUT` (?)

But as it is deprecated, I'm not sure if it will see some use. (libhidapi seems to rely on libusb0 if I saw that correctly?)

Querela, 2020-01-07:
Searched for libusb0 error codes and found only those for libusb-win32 v0.1 ...
-- https://sourceforge.net/p/libusb-win32/wiki/Documentation/
```c
/* Connection timed out */
#define ETRANSFER_TIMEDOUT 116
```

&rarr;

```python
errno.ETIMEDOUT = 110
    Connection timed out
```

Not sure what to check for. Need some feedback from other users about the real return code.

commit 4134e911 (2020-01-30): Add USBTimeoutError

Timeout errors are generally handled differently from other USB errors; it makes sense to have a specific exception for them.

For backwards compatibility, USBTimeoutError subclasses USBError.

Fixes: #182 ("Only a single error class for different error categories")
closed 2020-01-30 by jonasmalacofilho