pyusb/pyusb #183: Resource Busy for USB CDC-UART Device

hduffddybz, 2017-05-15:
Hi all,
I am a newbie to libusb and pyusb. Recently I am trying to use pyusb to communicate between my PC and board with NXP LPC 1837 processor. Although this post:https://github.com/walac/pyusb/issues/76 gives some solutions with similar questions, I still cannot solve it. Can anyone help me?

OS: Ubuntu 16.04
libusb: libusb-1.0-0
pyusb: latest git version

I used the example downloaded from https://github.com/walac/pyusb/blob/master/docs/tutorial.rst, to be specific, code from "Let's get it started" section. However when I run this script, it always gets the "Resource Busy error". In the following, I listed the log produced by this example.

`2017-05-15 16:25:29,780 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 1e05db0 at 7fb553382f50>)
2017-05-15 16:25:29,783 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2017-05-15 16:25:29,783 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2017-05-15 16:25:29,783 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb553396710>)
2017-05-15 16:25:29,783 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb553396850>)
2017-05-15 16:25:29,784 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb553396850>, 0)
2017-05-15 16:25:29,784 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x7fb553396850>)
2017-05-15 16:25:29,784 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x7fb553a32c90>, 1)
Traceback (most recent call last):
  File "test.py", line 13, in <module>
    dev.set_configuration()
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python2.7/dist-packages/usb/_debug.py", line 60, in do_trace
    return f(*args, **named_args)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 794, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
2017-05-15 16:25:29,787 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x7fb553a32c90>)
2017-05-15 16:25:29,787 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
`
It should be noticed that I tried to use USB CDC-UART protocol to communicate between PC and target board.

The following lists information about my target usb device extracted from "lsusb" command:
`Bus 002 Device 018: ID 1fc9:0083 NXP Semiconductors 
  93 Device Descriptor:
  94   bLength                18
  95   bDescriptorType         1
  96   bcdUSB               2.00
  97   bDeviceClass          239 Miscellaneous Device
  98   bDeviceSubClass         2 ?
  99   bDeviceProtocol         1 Interface Association
 100   bMaxPacketSize0        64
 101   idVendor           0x1fc9 NXP Semiconductors
 102   idProduct          0x0083 
 103   bcdDevice            1.00
 104   iManufacturer           1 NXP
 105   iProduct                2 VCOM Port
 106   iSerial                 3 NXP-77
 107   bNumConfigurations      1
 108   Configuration Descriptor:
 109     bLength                 9
 110     bDescriptorType         2
 111     wTotalLength           75
 112     bNumInterfaces          2
 113     bConfigurationValue     1
 114     iConfiguration          0 
 115     bmAttributes         0xc0
 116       Self Powered
 117     MaxPower              500mA
 118     Interface Association:
 119       bLength                 8
 120       bDescriptorType        11
 121       bFirstInterface         0
 122       bInterfaceCount         2
 123       bFunctionClass          2 Communications
 124       bFunctionSubClass       2 Abstract (modem)
 125       bFunctionProtocol       0 None
 126       iFunction               4 VCOM
 127     Interface Descriptor:
 128       bLength                 9
 129       bDescriptorType         4
 130       bInterfaceNumber        0
 131       bAlternateSetting       0
 132       bNumEndpoints           1
 133       bInterfaceClass         2 Communications
 134       bInterfaceSubClass      2 Abstract (modem)
 135       bInterfaceProtocol      0 None
 136       iInterface              4 VCOM
 CDC Header:
 138         bcdCDC               1.10
 139       CDC Call Management:
 140         bmCapabilities       0x01
 141           call management
 142         bDataInterface          1
 143       CDC ACM:
 144         bmCapabilities       0x02
 145           line coding and serial state
 146       CDC Union:
 147         bMasterInterface        0
 148         bSlaveInterface         1 
 149       Endpoint Descriptor:
 150         bLength                 7
 151         bDescriptorType         5
 152         bEndpointAddress     0x83  EP 3 IN
 153         bmAttributes            3
 154           Transfer Type            Interrupt
 155           Synch Type               None
 156           Usage Type               Data
 157         wMaxPacketSize     0x0010  1x 16 bytes
 158         bInterval               2
 159     Interface Descriptor:
 160       bLength                 9
 161       bDescriptorType         4
 162       bInterfaceNumber        1
 163       bAlternateSetting       0
 164       bNumEndpoints           2
 165       bInterfaceClass        10 CDC Data
 166       bInterfaceSubClass      0 Unused
 167       bInterfaceProtocol      0 
 168       iInterface              4 VCOM
 169       Endpoint Descriptor:
 170         bLength                 7
 171         bDescriptorType         5
 172         bEndpointAddress     0x01  EP 1 OUT
 173         bmAttributes            2
 174           Transfer Type            Bulk
 175           Synch Type               None
 176           Usage Type               Data
 177         wMaxPacketSize     0x0040  1x 64 bytes
 178         bInterval               0
 179       Endpoint Descriptor:
 180         bLength                 7
 181      bDescriptorType         5
 182         bEndpointAddress     0x81  EP 1 IN
 183         bmAttributes            2
 184           Transfer Type            Bulk
 185           Synch Type               None
 186           Usage Type               Data
 187         wMaxPacketSize     0x0040  1x 64 bytes
 188         bInterval               0
 189 Device Qualifier (for other device speed):
 190   bLength                10
 191   bDescriptorType         6
 192   bcdUSB               2.00
 193   bDeviceClass            0 (Defined at Interface level)
 194   bDeviceSubClass         0 
 195   bDeviceProtocol         0 
 196   bMaxPacketSize0        64
 197   bNumConfigurations      1
 198 Device Status:     0x0001
 199   Self Powered
`
Any ideas on how to solve this problem would be welcomed.

Thanks in advance for your help.

cioma, 2017-05-20:
Perhaps there is a Linux driver that gets attached to this device? If yes then you'd need to deattach it. All relevant functions are available in pyusb

mcuee, 2018-08-07:
Closing old issues as the OP has not come back with the reply for a long time. The answer is also clear enough.

closed 2018-08-07 by mcuee