pyusb/pyusb #184: The show_devices from my usb hardware - Windows 10 O.S.

catafest, 2017-05-26:
I install the libusb-win32-devel-filter-1.2.6.0.exe. 
I try to detect my usb mouse from usb devices with:
```
C:\Python27>python.exe
Python 2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 20:42:59) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
>>> from usb import *
>>> print usb.core.show_devices()
```
How to detect under windows and use it with pyusb and Windows 10 ?

eucliwoodhell, 2017-06-08:
Lib32 has a device test, check. if it is empty lib, execute Inf wizard and install device on lib.

mcuee, 2018-04-11:
And normally you do not want to use pyusb or libusb with a mouse or keyboard where Windows limit the access.

mcuee, 2018-08-07:
Closing old issues as the OP has not come back with the reply for a long time.

closed 2018-08-07 by mcuee