pyusb/pyusb #186: Errors on windows with the latest version of libusb-1.0

chibill, 2017-07-07:
Firstly when following the titorial code for custom_match 
if I try scanning a class of 8 or 3 (Basically any class) it returns a generator that when I loop thru it errors with 

> > Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "C:\Python27\lib\site-packages\usb\core.py", line 1249, in device_iter
>     if _interop._all(tests) and (custom_match is None or custom_match(d)):
>   File "<stdin>", line 10, in __call__
>   File "C:\Python27\lib\site-packages\usb\core.py", line 1094, in __iter__
>     yield Configuration(self, i)
>   File "C:\Python27\lib\site-packages\usb\core.py", line 586, in __init__
>     configuration
>   File "C:\Python27\lib\site-packages\usb\backend\libusb1.py", line 757, in get_configuration_descriptor
>     config, byref(cfg)))
>   File "C:\Python27\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
>     raise USBError(_strerror(ret), ret, _libusb_errno[ret])
> usb.core.USBError: [Errno 2] Entity not found

chibill, 2017-07-07:
Secondly I am getting not implemented errors while trying to use get_configuration and set_configuration on devices.

chibill, 2017-08-03:
Any help?

walac, 2017-08-03:
Sorry for the long delay, could you please post your code?

chibill, 2017-08-03:
`import usb.core
import usb.util
import sys

class find_class(object):
    def __init__(self, class_):
        self._class = class_
    def __call__(self, device):
        # first, let's check the device
        if device.bDeviceClass == self._class:
            return True
        # ok, transverse all devices to find an
        # interface that matches our class
        for cfg in device:
            # find_descriptor: what's it?
            intf = usb.util.find_descriptor(
                                        cfg,
                                        bInterfaceClass=self._class
                                )
            if intf is not None:
                return True

        return False

devices = usb.core.find(find_all=1, custom_match=find_class(3))

for x in devices:
    print x`

A very slightly modified version of code from https://github.com/walac/pyusb/blob/master/docs/tutorial.rst#where-are-you

chibill, 2017-08-03:
This problem also affects other libraries that use pyusb that I have tried.  So I am pretty sure its not a usage problem.

chibill, 2017-08-03:
Also I am on Windows 10
with Python 2.7.13 64-Bit

chibill, 2017-08-05:
So any ideas?

walac, 2017-08-06:
Hrm, could you please set `PYUSB_ERROR=debug` and `LIBUSB_DEBUG=4` environment variables and post the logs?

chibill, 2017-08-06:
Where would these logs be? (I set those but no new error messages show up)

chibill, 2017-08-06:
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00001650] libusb: debug [libusb_init] created default context
[ 0.001505] [00001650] libusb: debug [libusb_init] libusb v1.0.21.11156
[ 0.002507] [00001650] libusb: debug [windows_init] Windows 8 64-bit
[ 0.003008] [00001650] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.004011] [00001650] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.004011] [00001650] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.005515] [00001650] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.005515] [00001650] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.005515] [00001650] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.006016] [00001650] libusb: debug [windows_init_clock] hires timer available (Frequency: 2531250 Hz)
[ 0.006016] [00001650] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.007019] [00001650] libusb: debug [htab_create] using 1021 entries hash table
[ 0.007019] [00001650] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.007520] [00001650] libusb: debug [libusb_get_device_list]
[ 0.008523] [00001650] libusb: debug [windows_get_device_list] allocating new device for session [3E]
[ 0.009024] [00001650] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009024] [00001650] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009526] [00001650] libusb: debug [windows_get_device_list] allocating new device for session [8A]
[ 0.010027] [00001650] libusb: debug [windows_get_device_list] allocating new device for session [22A]
[ 0.010027] [00001650] libusb: debug [init_device] got bus number from ancestor #2
[ 0.010528] [00001650] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.011029] [00001650] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
[ 0.013534] [00001650] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 10): '\\.\USB#VID_8087&PID_0A2A#5&2A2FDE03&0&10'
[ 0.014537] [00001650] libusb: debug [windows_get_device_list] allocating new device for session [51]
[ 0.014537] [00001650] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.015039] [00001650] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1404 bytes)
[ 0.015540] [00001650] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 11): '\\.\USB#VID_04F2&PID_B521#200901010001'
[ 0.016041] [00001650] libusb: debug [windows_get_device_list] found existing device for session [8A] (1.0)
[ 0.016542] [00001650] libusb: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&DCB6C4C&0&0'
[ 0.017044] [00001650] libusb: debug [windows_get_device_list] allocating new device for session [3F9]
[ 0.017044] [00001650] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.017545] [00001650] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.018046] [00001650] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 7): '\\.\USB#VID_1770&PID_FF00#MSI_EPF_USB'
[ 0.019550] [00001650] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.019550] [00001650] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.020053] [00001650] libusb: debug [windows_get_device_list] found existing device for session [51] (1.3)
[ 0.020553] [00001650] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.020553] [00001650] libusb: debug [get_api_type] matched driver name against HID API
[ 0.021054] [00001650] libusb: debug [windows_get_device_list] found existing device for session [3F9] (1.2)
[ 0.021557] [00001650] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.022057] [00001650] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.022558] [00001650] libusb: debug [windows_get_device_list] found existing device for session [22A] (1.3)
[ 0.023060] [00001650] libusb: debug [windows_get_device_list] setting HID interface for [3F9]:
[ 0.023060] [00001650] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_1770&PID_FF00#6&2B03053&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.023561] [00001650] libusb: debug [windows_get_device_list] unlisted ancestor for '\\.\HID#ENE0110#6&6B5FD3D&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.024062] [00001650] libusb: debug [libusb_get_device_descriptor]
[ 0.024564] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.024564] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.025065] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.025566] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.026068] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.026569] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.029577] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.030078] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.030581] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.031081] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.031582] [00001650] libusb: debug [libusb_get_device_descriptor]
[ 0.032084] [00001650] libusb: debug [libusb_get_config_descriptor] index 0
[... cut ...]
  File "C:\Python27\lib\site-packages\usb\core.py", line 1094, in __iter__
    yield Configuration(self, i)
  File "C:\Python27\lib\site-packages\usb\core.py", line 586, in __init__
    configuration
  File "C:\Python27\lib\site-packages\usb\backend\libusb1.py", line 757, in get_configuration_descriptor
    config, byref(cfg)))
  File "C:\Python27\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found
```

chibill, 2017-08-06:
this is using the same code I posted before.

walac, 2017-08-06:
Ops, sorry, the environment variable name is `PYUSB_DEBUG=debug` not `PYUSB_ERROR`. Could you please run again?

chibill, 2017-08-06:
```
C:\Users\chibi\Desktop>python test.py
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [0000232c] libusb: debug [libusb_init] created default context
[ 0.000000] [0000232c] libusb: debug [libusb_init] libusb v1.0.21.11156
[ 0.000000] [0000232c] libusb: debug [windows_init] Windows 8 64-bit
[ 0.000000] [0000232c] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.031256] [0000232c] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.031256] [0000232c] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.031256] [0000232c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.031256] [0000232c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.046880] [0000232c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.046880] [0000232c] libusb: debug [windows_init_clock] hires timer available (Frequency: 2531250 Hz)
[ 0.046880] [0000232c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.046880] [0000232c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.046880] [0000232c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.046880] [0000232c] libusb: debug [libusb_get_device_list]
[ 0.046880] [0000232c] libusb: debug [windows_get_device_list] allocating new device for session [3E]
[ 0.062490] [0000232c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.062490] [0000232c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.062490] [0000232c] libusb: debug [windows_get_device_list] allocating new device for session [8A]
[ 0.062490] [0000232c] libusb: debug [windows_get_device_list] allocating new device for session [22A]
[ 0.062490] [0000232c] libusb: debug [init_device] got bus number from ancestor #2
[ 0.062490] [0000232c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.062490] [0000232c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
[ 0.062490] [0000232c] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 10): '\\.\USB#VID_8087&PID_0A2A#5&2A2FDE03&0&10'
[ 0.062490] [0000232c] libusb: debug [windows_get_device_list] allocating new device for session [51]
[ 0.062490] [0000232c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.062490] [0000232c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1404 bytes)
[ 0.062490] [0000232c] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 11): '\\.\USB#VID_04F2&PID_B521#200901010001'
[ 0.062490] [0000232c] libusb: debug [windows_get_device_list] found existing device for session [8A] (1.0)
[ 0.062490] [0000232c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&DCB6C4C&0&0'
[ 0.062490] [0000232c] libusb: debug [windows_get_device_list] allocating new device for session [3F9]
[ 0.062490] [0000232c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.062490] [0000232c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.062490] [0000232c] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 7): '\\.\USB#VID_1770&PID_FF00#MSI_EPF_USB'
[ 0.078115] [0000232c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.078115] [0000232c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.078115] [0000232c] libusb: debug [windows_get_device_list] found existing device for session [51] (1.3)
[ 0.078115] [0000232c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.078115] [0000232c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.078115] [0000232c] libusb: debug [windows_get_device_list] found existing device for session [3F9] (1.2)
[ 0.078115] [0000232c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.078115] [0000232c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.078115] [0000232c] libusb: debug [windows_get_device_list] found existing device for session [22A] (1.3)
[ 0.078115] [0000232c] libusb: debug [windows_get_device_list] setting HID interface for [3F9]:
[ 0.078115] [0000232c] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_1770&PID_FF00#6&2B03053&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.078115] [0000232c] libusb: debug [windows_get_device_list] unlisted ancestor for '\\.\HID#ENE0110#6&6B5FD3D&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.078115] [0000232c] libusb: debug [libusb_get_device_descriptor]
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.078115] [0000232c] libusb: debug [libusb_get_device_descriptor]
[... cut ...]
[ 0.124995] [0000232c] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.124995] [0000232c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.124995] [0000232c] libusb: debug [libusb_exit]
[ 0.124995] [0000232c] libusb: debug [libusb_exit] destroying default context
[ 0.124995] [0000232c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.124995] [0000201c] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

Thats the output (Notice this time i just put the code in a script...)

chibill, 2017-08-06:
Sort of looks like it destroyed the ref or something even tho it handed me a generator for is..

walac, 2017-08-07:
So, it feels like PYUSB_DEBUG is not defined. You should see something like [this](https://pastebin.mozilla.org/9029056).

fwiw, I tested on Mac OSX and the script worked successfully. This might be indeed a bug in libusb for Windows.

chibill, 2017-08-07:
I will try manually telling it that it's defined. (Going to remove where it gets the value with just the value)

chibill, 2017-08-07:
```
2017-08-07 10:25:22,730 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\Program Files\rtlsdr\libusb-1.0.dll', handle da0b0000 at 30d2278>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00002d60] libusb: debug [libusb_init] created default context
[ 0.000000] [00002d60] libusb: debug [libusb_init] libusb v1.0.21.11156
[ 0.000000] [00002d60] libusb: debug [windows_init] Windows 8 64-bit
[ 0.000501] [00002d60] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.048630] [00002d60] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.048630] [00002d60] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.049632] [00002d60] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.049632] [00002d60] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.062667] [00002d60] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.063169] [00002d60] libusb: debug [windows_init_clock] hires timer available (Frequency: 2531247 Hz)
[ 0.063169] [00002d60] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.063669] [00002d60] libusb: debug [htab_create] using 1021 entries hash table
[ 0.063669] [00002d60] libusb: debug [usbi_add_pollfd] add fd 0 events 1
2017-08-07 10:25:22,796 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2017-08-07 10:25:22,796 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.064171] [00002d60] libusb: debug [libusb_get_device_list]
[ 0.064673] [00002d60] libusb: debug [windows_get_device_list] allocating new device for session [3E]
[ 0.064673] [00002d60] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.065173] [00002d60] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.065173] [00002d60] libusb: debug [windows_get_device_list] allocating new device for session [8A]
[ 0.065173] [00002d60] libusb: debug [windows_get_device_list] allocating new device for session [22A]
[ 0.067179] [00002d60] libusb: debug [init_device] got bus number from ancestor #2
[ 0.067680] [00002d60] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.069185] [00002d60] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
[ 0.069686] [00002d60] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 10): '\\.\USB#VID_8087&PID_0A2A#5&2A2FDE03&0&10'
[ 0.069686] [00002d60] libusb: debug [windows_get_device_list] allocating new device for session [51]
[ 0.070187] [00002d60] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.070688] [00002d60] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1404 bytes)
[ 0.070688] [00002d60] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 11): '\\.\USB#VID_04F2&PID_B521#200901010001'
[ 0.071189] [00002d60] libusb: debug [windows_get_device_list] found existing device for session [8A] (1.0)
[ 0.071189] [00002d60] libusb: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&DCB6C4C&0&0'
[ 0.071691] [00002d60] libusb: debug [windows_get_device_list] allocating new device for session [327]
[ 0.072192] [00002d60] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.072693] [00002d60] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.072693] [00002d60] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 3): '\\.\USB#VID_046D&PID_C246#5&2A2FDE03&0&3'
[ 0.073195] [00002d60] libusb: debug [windows_get_device_list] allocating new device for session [3F9]
[ 0.073195] [00002d60] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.073696] [00002d60] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.073696] [00002d60] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 7): '\\.\USB#VID_1770&PID_FF00#MSI_EPF_USB'
[ 0.074699] [00002d60] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.075200] [00002d60] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.075200] [00002d60] libusb: debug [windows_get_device_list] found existing device for session [51] (1.2)
[ 0.075701] [00002d60] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.092747] [00002d60] libusb: debug [get_api_type] matched driver name against HID API
[ 0.093249] [00002d60] libusb: debug [windows_get_device_list] found existing device for session [3F9] (1.3)
[ 0.093749] [00002d60] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.094251] [00002d60] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.094251] [00002d60] libusb: debug [windows_get_device_list] found existing device for session [22A] (1.3)
[ 0.094752] [00002d60] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.095253] [00002d60] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.095755] [00002d60] libusb: debug [windows_get_device_list] found existing device for session [327] (1.4)
[ 0.096257] [00002d60] libusb: debug [windows_get_device_list] setting HID interface for [3F9]:
[ 0.096757] [00002d60] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_1770&PID_FF00#6&2B03053&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[... cut ...]
[ 0.164996] [00002d60] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.165998] [00002d60] libusb: debug [libusb_unref_device] destroy device 1.0
2017-08-07 10:25:22,898 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.168004] [00002d60] libusb: debug [libusb_exit]
[ 0.168004] [00002d60] libusb: debug [libusb_exit] destroying default context
[ 0.168505] [00002d60] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.169006] [000011d4] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

chibill, 2017-08-11:
Any clues in the debug logs?

walac, 2017-08-11:
I don't have a Windows machine available atm to debug it, but it feels like a libusb bug (not 100% sure, however). This week has been busy so I didn't look deeply at it yet. Maybe you lack some permission?

chibill, 2017-08-11:
Well I am the only account on the computer and it is an administrative account so I don't think so... so not sure.

walac, 2017-08-11:
Hrm, how did you install `libusb`? Could you please try with libusb-win32?

chibill, 2017-08-11:
? I installed what you have linked on the read what. (Granted the 64-bit version so python could actually load it)

chibill, 2017-08-11:
Also is libusb not the same as libusb-win32?
And does libusb-win32 have a 64-bit version?

chibill, 2017-08-11:
Sorry if I sounded rude.

walac, 2017-08-11:
> ? I installed what you have linked on the read what. (Granted the 64-bit version so python could actually load it)

Sorry, too much in my head, forgot it :)

> Also is libusb not the same as libusb-win32?

libusb-win32 is the implementation of libusb 0.1 API on Windows. libusb-1.0 is a entirely new library. PyUSB support both. You can install it with zadig too.

> And does libusb-win32 have a 64-bit version?

afaik, yes.

> Sorry if I sounded rude.

No worries :-)

chibill, 2017-08-11:
Well I looked at the download site for libusb-win32 there's no 64bit binaries. :( and I don't see how using a utility to install a driver will help. When I have no device I want to force to use it. (Unless you can install the .dll thru it just with out hurting a device by changing its driver)

chibill, 2017-08-13:
If I force it to use libusb-win32 this is all the output I get.

`2017-08-13 10:12:15,088 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
`

Which is odd when I have debugging on for pyusb...

And it still can't see my three devices I know exist on usb. (Granted they don't have libusb as the driver but that should not matter...)

walac, 2017-08-13:
> And it still can't see my three devices I know exist on usb. (Granted they don't have libusb as the driver but that should not matter...)

libusb will only shows the devices for which it has its driver loaded for... Which kind of device are you trying to talk to?

chibill, 2017-08-13:
So libusb-win32 can only see devices that I remove the default drive one?
That seems odd. But okay. So there is no way to actually make pycanon work anymore? (It uses pyusb to talk to a Canon camera. And it uses the default driver)

chibill, 2017-08-13:
So unless I mess up the software for my device to experiment with it in python.... :(

walac, 2017-08-14:
> So libusb-win32 can only see devices that I remove the default drive one? That seems odd. But okay. So there is no way to actually make canon-remote work anymore? (It uses pyusb to talk to a Canon camera. And it uses the default driver)

Could you please link me to the project? I am installing a windows VM to try to simulate the issue, but not having much free time lately.

chibill, 2017-08-14:
https://pythonhosted.org/canon-remote/

chibill, 2017-08-26:
Well also testing on devices that work with out messing up the driver. (According to libusb people) 
I also get a lot of "NotImplementedError: Operation not supported or unimplemented on this platform"
just following the tutorial. (ex. dev.set_configuration() or dev.get_active_configuration)

edmont, 2018-03-01:
Same problem here on Windows 8, Python 3.6.4 64-bit.

mcuee, 2018-04-07:
For libusb 1.0 Windows backend, you can use usbdk backend to keep the existing driver.

mcuee, 2018-04-07:
And make sure you use latest version of libusb under Windows (1.0.22).

mcuee, 2018-04-07:
And to diagnose whether this is a libusb problem or pyusb problem, you can write a simple C program like the one by the OP to see if you can reproduce the issue or not. If there is no issue with that, then the problem is with pyusb.

JamesTheAwesomeDude, 2018-07-25:
I am also getting this exact issue with libusb 1.0.22 installed as per [@KM4YRI's instructions](https://github.com/pyusb/pyusb/issues/120#issuecomment-322058585) (i.e., dropped `libusb-1.0.22.7z/MS64/dll/libusb-1.0.dll` into `C:\Windows\System32\`

The following transcript from an administrator shell:

````
PS C:\WINDOWS\system32> python3
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.util,usb.core
>>> dev=usb.core.find(idVendor=0x04d8, idProduct=0xc002)
>>> dev.reset()
>>> dev.set_configuration()
>>> usb.util.claim_interface(dev,0)
>>> dev.set_interface_altsetting(0)
>>> dev.write(1,b'\x91\x7f')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\jedingto\AppData\Roaming\Python\Python36\site-packages\usb\core.py", line 948, in write
    self.__get_timeout(timeout)
  File "C:\Users\jedingto\AppData\Roaming\Python\Python36\site-packages\usb\backend\libusb1.py", line 842, in intr_write
    timeout)
  File "C:\Users\jedingto\AppData\Roaming\Python\Python36\site-packages\usb\backend\libusb1.py", line 920, in __write
    _check(retval)
  File "C:\Users\jedingto\AppData\Roaming\Python\Python36\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
>>>
````

Am currently running Windows 10

> Version 1709 (OS Build 16299.431)

JamesTheAwesomeDude, 2018-07-31:
Looks like an OK workaround for now is to use [libusb-win32@SF](http://libusb-win32.sourceforge.net/)

If you can't wrangle the INF installation into working due to its being "unsigned", the "simple" manual installation is:

1. Drop `libusb-win32-bin-1.2.6.0/bin/${ARCH}/libusb0.{dll,sys}` into `C:\Windows\System32\` and `C:\Windows\System32\drivers\` (respectively)
2. Use the appropriate `install-filter-win.exe` to attach libusb0 to the PicoLCD device (`vid:04d8`)

However, from the website (emphasis original):

> **libusb-win32 is now in Bug-Fix-Only maintenance mode. It is recommended new project should migrate to libusb-1.0 API and use libusb Windows (http://libusb.info) instead.**

(It immediately follows that up with:

> **For Windows only project, you can also look at libusbK below (http://libusbk.sourceforge.net/UsbK3/index.html)**

Although the libusbK website notes:

> **Bindings for other languages**
> The main libusbK library is written in standard C, but official bindings for the libusbK standard C library are provided for the folllowing languages:
> 1. C#
> 2. VB.NET
> 3. Pascal (Compliments of Ekkehard Domning)
> 4. BOO (http://boo.codehaus.org/)

so while libusbK is a promising replacement backend, it's not a completely ready-made solution to this)

JamesTheAwesomeDude, 2018-07-31:
A thought:

From [`libusb1`'s wiki](https://github.com/libusb/libusb/wiki#supported-environments):

> * [Windows](https://github.com/libusb/libusb/wiki/Windows) [**status**: Stable]
>   * [MinGW](http://www.mingw.org/) (32 bit) or [MinGW-w64](http://mingw-w64.sourceforge.net/). To compile a library that is both 32 and 64 bit compatible, you should use a version of MinGW-w64 that supports both `-m32` and `-m64` (a.k.a. "multilib"). A pre-built multilib version can be downloaded with TDM64.
>   * Microsoft Visual C++ (Visual Studio).
>   * [Windows DDK build environment](http://www.microsoft.com/downloads/details.aspx?displaylang=en&FamilyID=36a2630f-5d56-43b5-b996-7633f2ec14ff)
>   * Device driver support: WinUSB, HID, libusb-win32 (libusb0.sys, not recommended), libusbK, usbdk.

Taking note of that last line, does libusb1 need one of these "backend drivers" to work?

mcuee, 2018-08-01:
@JamesTheAwesomeDude I think your issue is different. Your error message is "usb.core.USBError: [Errno 5] Input/Output Error", which means most likely the device does not understand the command "dev.write(1,b'\x91\x7f')". You need to understand the communication protocol in order to use pyusb/libusb.

 You may want to start a new thread.

mcuee, 2018-08-01:
@JamesTheAwesomeDude I am one of the admins of libusb, libusb-win32 and libusbk. If you are not clear about them, you can ask in the libusb mailing list.

libusb-win32 is mainly for existing users who are already using libusb-win32, if it works, no need to change. And it actually works for most of the use cases.

libusbK is Windows only and pyusb does not really support libusbk.

libusb (libusb-1.0 API) is the preferred library to go to for cross-platform project. It has several backend under Windows, WinUSB (most mature), libusbK and usbdk.

mcuee, 2018-08-01:
@wgaylord BTW, libusb-win32 has both 32/64bit driver/dll. The easiest way is to use libusbK installer to install latest version of libusb-win32/libusbk binary.

And you are right, I can reproduce your issue under Windows with libusb 1.0.21/1.0.22 and latest git version of pyusb.

AndCycle, 2018-10-15:
things just worse, libusb 1.0.21 broken under windows 10 1809
```
'USBError' object has no attribute 'reason'

Traceback (most recent call last):
  File ".\start_gui.py", line 166, in start_proxyagent
    ykaccess.ret_serial()
  File "D:\usbproj\ykaccess.py", line 214, in ret_serial
    return YubiAccess.ret_serial()
  File "D:\usbproj\ykaccess.py", line 51, in ret_serial
    return key.serial()
  File "D:\usbproj\ykaccess.py", line 122, in serial
    YK = self.ret_yk()
  File "D:\usbproj\ykaccess.py", line 117, in ret_yk
    YK = yubico.find_yubikey()
  File "D:\usbproj\runtime36\lib\site-packages\yubico\yubikey.py", line 53, in find_key
    hid_device = YubiKeyHIDDevice(debug, skip)
  File "D:\usbproj\runtime36\lib\site-packages\yubico\yubikey_usb_hid.py", line 123, in __init__
    if not self._open(skip):
  File "D:\usbproj\runtime36\lib\site-packages\yubico\yubikey_usb_hid.py", line 338, in _open
    self._usb_handle.claimInterface(self._usb_int)
  File "D:\usbproj\runtime36\lib\site-packages\usb\legacy.py", line 230, in claimInterface
    util.claim_interface(self.dev, interface)
  File "D:\usbproj\runtime36\lib\site-packages\usb\util.py", line 205, in claim_interface
    device._ctx.managed_claim_interface(device, interface)
  File "D:\usbproj\runtime36\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "D:\usbproj\runtime36\lib\site-packages\usb\core.py", line 159, in managed_claim_interface
    self.managed_open()
  File "D:\usbproj\runtime36\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "D:\usbproj\runtime36\lib\site-packages\usb\core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "D:\usbproj\runtime36\lib\site-packages\usb\backend\libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "D:\usbproj\runtime36\lib\site-packages\usb\backend\libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "D:\usbproj\runtime36\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)
```

libusb 1.0.22 got OS:Error exception with pyusb,
my app just forced into a corner.

mcuee, 2019-12-07:
I can reproduce the issue with MSYS2 MinGW-w64, latest pyusb, latest libusb git.

> import usb.core
import usb.util
import sys

class find_class(object):
    def __init__(self, class_):
        self._class = class_
    def __call__(self, device):
        # first, let's check the device
        if device.bDeviceClass == self._class:
            return True
        # ok, transverse all devices to find an
        # interface that matches our class
        for cfg in device:
            # find_descriptor: what's it?
            intf = usb.util.find_descriptor(
                                        cfg,
                                        bInterfaceClass=self._class
                                )
            if intf is not None:
                return True

        return False

devices = usb.core.find(find_all=1, custom_match=find_class(3))

for x in devices:
    print x

mcuee, 2019-12-07:
@AndCycle yubikey.py seems to indicate that is an HID device, and your libusb-1.0.21 log is a bit different, can you try latest libusb-1.0.23? And take note you may want to use HIDAPI and not libusb for Windows HID device. HIDAPI has python bindings. HIDAPI is now also under github libusb organization.

jonasmalacofilho, 2020-01-05:
I think this is an libusb issue...

I can reproduce it on Windows, but only on `1b21:2142` (ASMedia USB 3.1 eXtensible Host Controller) and `8086:a2af` (Intel(R) USB 3.0 eXtensible Host Controller), both of which are USB root hubs which Windows reports as PCI devices, not USB ones.

Additionally, for each of these libusb returns to PyUSB that there is one configuration... so when it later returns `LIBUSB_ERROR_NOT_FOUND`when fetching config 0, it reinforces my suspicion that it's a libusb issue.

<details><summary>Expand for my slightly modified script</summary>

```py
import usb.core
import usb.util
import sys

class find_class(object):
    def __init__(self, class_):
        self._class = class_
    def __call__(self, device):
        # first, let's check the device
        if device.bDeviceClass == self._class:
            return True
        # ok, transverse all devices to find an
        # interface that matches our class
        print(f'{device.idVendor:04x}:{device.idProduct:04x} @ {device.bus}:{device.address}: bNumConfigurations={device.bNumConfigurations}')
        try:
            for cfg in device:
                # find_descriptor: what's it?
                intf = usb.util.find_descriptor(cfg, bInterfaceClass=self._class)
                if intf is not None:
                    return True
        except Exception as err:
            print(err)

        return False

devices = usb.core.find(find_all=1, custom_match=find_class(3))

for x in devices:
    # print(x)
    pass
```
</details>

---

Note that I haven't made I a C reproduceable example yet (don't have a development environment set up on this Windows machine), so I could be wrong...

mcuee, 2020-01-06:
Great that you can reproduce this issue. 

As for USB configuration, Configuration 0 is not a valid configuration, it actually means that the device is not configured (not ready or not present).

jonasmalacofilho, 2020-01-07:
My bad, I meant to say configuration *index* 0... which should be valid if `bNumConfigurations >= 1`, right?

This is passed to `libusb_get_config_descriptor`, which then fails on two root hubs on Windows (but succeeds on Linux, on the same machine).  The only difference seems to be that on Linux libusb doesn't return these two root hubs.

Wei1234c, 2020-01-14:
My two cents:
- There is a PCI device ([USB 3.0 xHCI Controller](https://devicehunt.com/view/type/pci/vendor/8086/device/A12F) in my case) among the devices listed by libusb.
- Exception occures when try to get_configuration_descriptor() from a PCI device.

I modify `Python/Lib/site-packages/usb/backend/libusb1.py` as :
```
    @methodtrace(_logger)
    def get_configuration_descriptor(self, dev, config):
        cfg = POINTER(_libusb_config_descriptor)()

        # debug code start
        # to show which device cause the problem. read the idVendor/idProduct
        dev_desc = _libusb_device_descriptor()
        _check(self.lib.libusb_get_device_descriptor(dev.devid, byref(dev_desc)))

        ret = self.lib.libusb_get_config_descriptor(
            dev.devid,
            config, byref(cfg))

        print(ret, ': ', hex(dev_desc.idVendor), hex(dev_desc.idProduct))

        # debug code end, break if ret < 0

        _check(self.lib.libusb_get_config_descriptor(
                dev.devid,
                config, byref(cfg)))
        config_desc = _ConfigDescriptor(cfg)
        config_desc.extra_descriptors = (
                config_desc.extra[:config_desc.extra_length])
        return config_desc

```

and return :
```
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x8087 0xa2a
0 :  0x46d 0xc52b
0 :  0x46d 0xc52b
-5 :  0x8086 0xa12f
```

jonasmalacofilho, 2020-02-01:
I reproduced the problem without any dependencies to PyUSB, and reported it upstream (link above).

As for implementing a workaround in PyUSB, I think that we should wait and see what's indeed the problem, on the libusb level.

In particular, I think that stopping `find` from listing the root hubs (which is what the proposed patch does) might be too drastic (at the moment, at least).

mcuee, 2020-07-01:
Maybe it is a good idea to skip root hubs for pyusb. You can not do much with root hubs with pyusb and even libusb. But for libusb it is more for legacy reasons.

mcuee, 2020-12-06:
BTW, libusb issue #688 has been closed and patch merged.
https://github.com/libusb/libusb/issues/688

closed 2020-12-08 by jonasmalacofilho