pyusb/pyusb #188: Problems to connect to NXT brick

Va2sili, 2017-07-26:
Hello, Im using nxt-python 1.2.0, PyBluez 0.22 and PyUsb 1.0.0 on python 2.7 on Windows 10.
I tried to connect to my NXT brick with bluetooth and with usb but none method worked.
-with bluetooth, I get the following error : 'IOError: a connection attempt failed because the brick didnt reply fast enough" when I execute one of the example program. However, python still manages to give the bluetooth adress of the brick (or something else I dont know) using 'host' even if it's not turned on ! 'nxt.locator.find_one_brick()' gives '<nxt.bluesock.BlueSock object at 0x04481630>'
-with usb, I get : 'USBError: [Errno None] libusb0-dll:err [control_msg] sending control message failed, win error: The specific file cant be found.\r\n\n\n'. 'nxt.locator.find_one_brick()' gives <nxt.usbsock.USBSock object at 0x0750ADF0>
Can you please help me to connect to the brick with at least one of the both methods. Thanks

chibill, 2017-08-03:
I don't think pyusb is maintained.... :(

Considering I have had an issue open for a month with no reply.

mcuee, 2017-10-07:
Maybe you want to try to libusb-1.0 and WinUSB instead of libusb-win32 and try again.

Va2sili, 2017-10-14:
Okay thank you I will try !

mcuee, 2018-08-07:
Closing old issues as the OP has not come back with the reply for a long time.

closed 2018-08-07 by mcuee