pyusb/pyusb #196: PyUSB failure to read from Keyboard or Flash Drive on Mac due to Permission Errors

UsersDoProgram, 2018-01-02:
When running the following code https://pastebin.com/6pcyGHKg that aims to read data from a keyboard or flash drive with PyUSB, Python 3.6.3 and a Mac on OSX 10.13. I get an error mentioning Insufficient Permissions. I have tried running the code with while logged in as root and I still get the same issue.
Traceback (most recent call last):
  File "usbReader.py", line 27, in <module>
    main(1)
  File "usbReader.py", line 24, in main
    data = dev.read(endpoint.bEndpointAddress,endpoint.wMaxPacketSize)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/core.py", line 975, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

mcuee, 2018-04-07:
You have to read libusb FAQ to understand that you will not be able to use libusb/pyusb under Mac OS X for those device bound to kernel driver.
https://github.com/libusb/libusb/wiki/FAQ#How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device

mcuee, 2018-08-07:
Close this for now.

closed 2018-08-07 by mcuee