pyusb/pyusb #198: ImportError: No module named 'usb.core'; 'usb' is not a package

DouglasLapsley, 2018-05-01:
Hi
I'm installed pyusb-1.0.2 on my raspberry pi (Jessie) with the following:

`pip3 install pyusb`

I import in python with...

`import usb.core`
or just
`import usb`

Either way I get the following error...

ImportError: No module named 'usb.core'; 'usb' is not a package

...when running the following from terminal:

`python3 testusb.py`

It doesn't seem to matter if I run it as sudo or not.

What's really strange is that I managed to get it working the other day, but then I turned on the pi a day later and boom it started failing like this again.

Is this is a bug?  Am I doing something wrong?

Thanks very much!

DouglasLapsley, 2018-05-09:
I'm such a donkey.  I had called the file usb.py so it was trying to import itself!  Renamed the file and everything works perfectly.

closed 2018-05-09 by DouglasLapsley

MetaFight, 2020-07-25:
I'm experiencing this issue.  What did you do to resolve it?

mxx-lxg, 2020-09-21:
Any fix for this? I'm having the same problem.

jonasmalacofilho, 2020-09-21:
It seems unlikely that you have the exact same setup as the OP.

Can you please open another issue and provide all relevant context, like OS type and version, Python version, PyUSB version and path (see `pip show`), the error message and the traceback?

Please also include the output of:

```
python -c "import usb; print(usb.__version__)"
```

P.S. Depending on your setup you may need to use `pip3` and `python3` or even specific paths to ensure the desired pip and Python versions are used.

mxx-lxg, 2020-09-21:
Nevermind. My problem was that my script was called "usb.py" so it tried to import itself.

yoyo115956, 2023-12-13:
> Nevermind. My problem was that my script was called "usb.py" so it tried to import itself.

Haha, me too. Anyway, thanks a lot!!!

redo33, 2024-12-21:
I'm very glad that I found this thread. 

Long live. Github !!